AlbrechtL/welle.io #1: ui_filereader-widget.h not found

andimik, 2017-02-08:
Seems in `src/input/rawfiles/rawfiles.h` you're still referring to `ui_filereader-widget.h` which is not existing.

Therefore, I cannot compile it.

If I comment out all lines (26, 48, 91, 138) pointing to rawfiles, then it works.

I'm owning an Airspy Mini and would still have the filereader (for sdr or raw files) integrated ...

andimik, 2017-02-08:
And the line `-DESTDIR		= ./linux-bin` has been deleted in the pro-file, so the created executable file is in the welle.io directory.

AlbrechtL, 2017-02-08:
I assume you are a developer so I can go a little bit more in detail.

I delete the SDR input drivers except rtlsdr because I only own a rtlsdr thus I only can test the rtlsdr. If you would like to add the airspy input again. Go ahead and sent me a pull request!
You can also maintain the rawfiles input. I prefer to configure the rawinput via the command line because this feature is only for developers.
I removed the widgets because I plan to removed them at all from all input drivers. The background is that the QML GUI is only working hardware accelerated on a Raspberry Pi if no references to QT widgets are included and thus welle.io can run without X.

So I invite you to work on this great program.

andimik, 2017-02-08:
Well, I'm not a developer ;-) but am in close contact with Jan having ideas for better usability. But I cannot program any code, just adapt existing lines ...

AlbrechtL, 2017-02-08:
OK. I will receive an airspy device in the next weeks. Then I will add the airspy support again. Please by patient.
Regarding the rawfiles support. This is not my highest priority.

AlbrechtL, 2017-02-17:
Fixed in commit 44816d7.
Use the following command line to use the raw file input.
`welle-io -D rawfile -F YourFile`

closed 2017-02-17 by AlbrechtL