AlbrechtL/welle.io #9: Open Tasks

AlbrechtL, 2017-02-23:
A list with open tasks con be found here: https://github.com/AlbrechtL/welle.io/wiki/Open-Tasks

mpbraendli, 2017-02-28:
Hi Albrecht,
a suggestion for the input devices: add support for SoapySDR!
https://github.com/pothosware/SoapySDR/wiki

AlbrechtL, 2017-02-28:
Thanks for the hint. I will add it to the open task list.

chrisdebian, 2017-03-16:
Hi, as per your aspiration, could you formally add 'Create Ubuntu Snap' (https://www.ubuntu.com/desktop/snappy) to the 'Open Task' list?

Thanks,

Chris
chris_debian
2E0FRU

AlbrechtL, 2017-03-16:
Thanks for the idea! I tried it once to create a Snap but I fail to pack QT into the Snap.
I added it to the task list. I hope that somebody takes this job!

andimik, 2017-03-23:
There is a nice github feature for formatting tasks:

https://help.github.com/articles/basic-writing-and-formatting-syntax/#task-lists

AlbrechtL, 2017-03-23:
Thanks for the hint.

AlbrechtL, 2017-11-24:
Closed to make cleanup the open issue list.

closed 2017-11-24 by AlbrechtL

AlbrechtL, 2018-01-23:
@joggele I don't know how to contact you so use this issue. We found your [rtl-dab](https://github.com/Joggele/rtl-dab) improvements and I invite you to collaborate with welle.io. We are highly looking for new developers that are improving the DAB SDR software and implement new features.