AlbrechtL/welle.io #14: DAB not working

AlbrechtL, 2017-03-14:
It seems that DAB is not working.
Tested with an odr-dabmod generated file.

whiskerp, 2017-03-15:
DAB+ seems to be OK byt not DAB.

AlbrechtL, 2017-03-15:
How did you try it? With real broadcasting or with [ODR](http://www.opendigitalradio.org)?
If possible can you provide a [RAW file](https://www.welle.io/devices/rawfile)?

whiskerp, 2017-03-15:
With real DAB.

sm0yyq, 2017-03-16:
Can say the same, DAB+ seems to work, but DAB sound's scrambled.

whiskerp, 2017-03-16:
Exactly. It is not decoding DAB properly and it just squeaks and bubbles.
DAB+ is fine.

AlbrechtL, 2017-03-16:
I know that DAB was working last (August 2016). But there where so much changes in the last 6 months  and this makes it hard to find the bug. And on my location I receive only DAB+.
I would be very happy if somebody provides a RAW file.

AlbrechtL, 2017-03-17:
OK. I found out that commit 17f692d destroys the DAB support. Does somebody see the reason?

AlbrechtL, 2017-03-22:
I found the problem. Please try again the latest version and report if the [fix](https://github.com/AlbrechtL/welle.io/commit/073840d29ef863db4c43265a3f0374229e63f8ac) works properly.

whiskerp, 2017-03-23:
Hi

 

I’m sorry but I don’t have a Windows build environment set up so I’ll have to wait until your next release.

 

Thanks

Peter

AlbrechtL, 2017-03-23:
The fix is included in this version: https://github.com/AlbrechtL/welle.io/releases/tag/V20170323

closed 2017-03-23 by AlbrechtL

tnadolecki, 2017-03-23:
I'll test now. 
In the meantime here is raw file containing DAB ,
http://matkaboska.sytes.net/s/tvascXIUIsC8UBe
Yup, finally learned how to record and play the right way :) 
Thank You 
Tom.

tnadolecki, 2017-03-23:
I know it's bit late but I also confirm it is working now in DAB.
I apreciate all help.

whiskerp, 2017-03-23:
Many thanks. DAB and DAB+ both working 😊