AlbrechtL/welle.io #16: Sometimes, scanner skips channel 5C

AndrejaKo, 2017-03-16:
So I have a multiplex on channel 5C. Sometimes, during scanning, the program will try to scan channel 5B for long time and afterwards completely skip channel 5C and move on to 5D. 
[image]

I am unable to reliably reproduce the issue, however it happens in around 30% of scans. 

P.S. Any hints on how to make this bug report more useful are welcome!

AlbrechtL, 2017-03-16:
Thanks for reporting. Yes, the channel scan is worth to improve. I added it to the [task list}(https://github.com/AlbrechtL/welle.io/wiki/Open-Tasks)

andimik, 2017-03-22:
I fixed the channel problem by editing the channels (= deleting those not in use) in the `DabConstants.cpp` file. But normally I use qt-dab and welle.io only offline having no DAB reception at home.

AlbrechtL, 2017-04-11:
The channel scan is improved now. Please try it again with 0eb4a3f.

closed 2017-04-11 by AlbrechtL