AlbrechtL/welle.io #17: Sound problem

zygmund2000, 2017-03-16:
Sound problem under linux Manjaro x64, after some time sound is very dropped.

AlbrechtL, 2017-03-16:
Can you explain the problem a little more in detail, please.
How is your signal quality? Are all three fields in the right top corner green? If no than you have a bad DAB reception.

zygmund2000, 2017-03-16:
My singnal is 20dB SNR and very good spectrum, afer start audio is ok but atere some time about 1-2min is not posiible to listen, for example qt-dab works without problems.

AlbrechtL, 2017-03-16:
Can you post the output from the command line?

qt-dab uses portaudio for the audio output while welle.io uses QT multimedia for the audio output. Obviously this makes a difference.

zygmund2000, 2017-03-16:
Audio: IdleState
Audio: StoppedState
Audio: ActiveState
Audio: ActiveState
Audio: IdleState
Audio: StoppedState
Audio: ActiveState
Audio: ActiveState
Audio: IdleState
Audio: StoppedState
Audio: ActiveState
Audio: ActiveState
Audio: IdleState
Audio: StoppedState
Audio: ActiveState
Audio: ActiveState
Audio: IdleState
Audio: StoppedState
Audio: ActiveState
Audio: ActiveState
Audio: IdleState
Audio: StoppedState
Audio: ActiveState
Audio: ActiveState
Audio: IdleState
Audio: StoppedState

AlbrechtL, 2017-03-16:
Which QT version do you use?

zygmund2000, 2017-03-17:
I installed from this link: https://www.qt.io/download-open-source/ and default checked 5.8

zygmund2000, 2017-03-17:
I'm using pulseaudio with default sample rate 48khz, this problem looks like audio data speed mismatch, eg sound card plays faster or slower then decoded from your app.

AlbrechtL, 2017-03-17:
How it sounds like? If you cards plays it faster or slower you should hear it.

zygmund2000, 2017-03-17:
LOL if difference is like 48000 or 48002 will you hear it?

AlbrechtL, 2017-03-19:
Can you compile the latest version and test it again, please?
I added some audio [code](e3c2e68) thats avoids that the sound cards stops playing.

BTW: Do you have a special sound card?

zygmund2000, 2017-03-20:
Still same problem, I don't have special sound card, just internal in laptop, I have other machine with different sound card but audio is also not continuous.

AlbrechtL, 2017-03-24:
Did you activate the expert view? Does it also happen with deactivated expert view?

AlbrechtL, 2017-03-26:
May it is the problem https://github.com/AlbrechtL/welle.io/issues/35. Please report if it is this issue!

AlbrechtL, 2017-03-28:
@zygmund2000: Thanks for reporting. I close this issue because the problem is explained in issue #35.

closed 2017-03-28 by AlbrechtL