AlbrechtL/welle.io #21: Freeze after channel scan with no result

AlbrechtL, 2017-03-17:
When search and no reception results, no rescan is possible and application freezes.

tnadolecki, 2017-03-23:
Does not seem to matter whether there were stations found or not. Freezes at every second scan attempt.

AlbrechtL, 2017-03-24:
Which operating system do you are using?

tnadolecki, 2017-03-24:
I've experienced this on windows and linux. I'm totally back on linux however now.

tnadolecki, 2017-03-24:
Please see the video I recorded for this purpose.
https://youtu.be/NVtLX1dz6TE

AlbrechtL, 2017-03-24:
It is possible that you post screen shots from this problem? The youtube videos doesn't exist.

tnadolecki, 2017-03-25:
https://www.youtube.com/watch?v=NVtLX1dz6TE
[https://www.youtube.com/watch?v=NVtLX1dz6TE](url)

AlbrechtL, 2017-03-25:
The  "No channel in index"  is not the problem. Can you tell me  a little bit more about your system, please (Linux version, QT version, RTL-SDR dongle, CPU)?

tnadolecki, 2017-03-25:
QT I only downloaded from their website 2 days ago. I don't know how to check the version, but it must be the latest.
RTL-SDR is nooelec with high stability TCXO.

0:  Realtek, RTL2838UHIDIR, SN: 00000001
Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner

CPU is DELL Laptop E6410 running Ubuntu 16.04.2 LTS

I was experiencing same behavior when in Windows 10. It would simply freeze also.

AlbrechtL, 2017-03-25:
I think it is a GUI problem. You have the same software configuration that I have. The GUI (QML) is hardware accelerated with the GPU and you experienced it with Ubuntu and Windows. Maybe it's some problem with the GPU and QML.

AlbrechtL, 2017-04-13:
Please try it again with version 52d06f6.

closed 2017-04-13 by AlbrechtL