AlbrechtL/welle.io #22: allow choosing audio device

ec1oud, 2017-03-19:
QAudioDeviceInfo::availableDevices() will give you a list; for each of those, put deviceName() into a combobox.  Let the user choose one, and then use this constructor: 

AudioOutput = new QAudioOutput(device, AudioFormat, this);

The combobox wouldn't take up too much space on the settings pane.

AlbrechtL, 2017-03-20:
Thanks for your idea.
Can you please explain your use case a little bit more in detail please?
From my point of view it is sufficient to choose the default sound card inside your OS and welle.io uses just the default sound card. If you would like to switch the sound card just set a new default sound card.

ec1oud, 2017-03-21:
I plugged in a USB DAC, so I want to use it; my motherboard sound card (the default) doesn't sound as good, and lately just doesn't work at all.  (I was experimenting with using an optical outboard DAC but that also stopped working.)  Linux sound configuration is too complicated - "just set a new default" is something I've always struggled with, for decades now, because the details are always changing, and different programs have different ideas of which sound system they ought to use (OSS, Also, pulseaudio, jack, nss, ESD, some KDE thing, ...).  Periodically it just breaks, and I end up doing without sound at all for a couple of years; especially for youtube videos, I just had to give up and use a mac for the last few years, because browsers don't have this sound card selector UI.  In the latest iteration, pulseaudio seems to be working well enough.  I wonder how long that will continue this time.  But configuration is still scattered to the winds - separate config files for pulseaudio and alsa, there are multiple pulseaudio control panels to choose from, multiple tray icons, mixers etc.  Until recently, I was actively avoiding pulseaudio, because it was taking some CPU time continuously, which meant my CPU fan would never idle down to the lowest speed.  (And for that matter, the kernel drivers and/or the hardware can often do some mixing of multiple streams anyway, so I'm still not sure why there is this idea that everybody needs a userspace mixer daemon running all the time.  And yet, sometimes with plain ALSA, I get that error with some programs that the sound device is busy and can't be used, even though nothing is playing.  Probably some badly-written software is locking it then.)  But that is fixed for now... fingers crossed.  But I see that pulseaudio's maximum bitrate is 48KHz, but my USB DAC can do higher rates, so I still think maybe something is lost in translation if I try to play 96KHz file, for example.

Even those who don't bother with separate DACs might want to switch back and forth between HDMI audio and motherboard audio, when it happens to work.

Pulseaudio also has a feature which can provide a virtual sound card which sends all the sound to all the devices at the same time, just in case you can't otherwise control the outputs well enough.  But that's a different device; not sure if it can be the default.

What if you want to use different speakers and headphones for different purposes?  Maybe send the output from radio programs like this to some speakers which are closer to your ears instead of the main music-oriented system, or something like that.

In summary, sound on Linux has been about a 20-year-long train wreck, and I have no faith whatsoever that it's quite done crashing and crumpling and morphing yet.  So I think every program with significant AV capabilities should provide a way to select a sound card, especially when Qt is providing such nice API for that.  VLC and Audacity and GQRX have nice UI for selecting the device, so those are the programs that I can rely on being able to use over the long term; the ones that don't, typically end up not working at some point.

AlbrechtL, 2017-03-27:
I understand your point Linux with audio has a lot of improvement potential.

But every feature that we add has to be tested and maintained. I also think that 99 % of the users are fine with default sound card. At the moment there are so many other open tasks that we don't have the capacity to add a new feature that only 1 % user uses.

closed 2017-03-27 by AlbrechtL