AlbrechtL/welle.io #23: [Build error] Unknown modules in Qt: qml quick charts multimedia

Wikinaut, 2017-03-20:
I followed your build instructions for Ubuntu but received the error after adding the `welle.io.pro file

`Unknown modules in Qt: qml quick charts multimedia`

Any idea how to fix this ?

System:

* Linux Mint 18.1 which is based on Ubuntu 16.04

Wikinaut, 2017-03-20:
fix 1: 

`sudo apt install qtdeclarative5-dev qtmultimedia5-dev`

AlbrechtL, 2017-03-21:
Please follow the compile [instructions for Ubuntu](https://github.com/AlbrechtL/welle.io#ubuntu-linux-1604-lts) and don't use the repository QT version because it is outdated.

closed 2017-03-21 by AlbrechtL

Wikinaut, 2017-03-21:
@AlbrechtL I did follow your instructions (after having purged the repository version). So QTcreator is showing Qt5.8 version including Charts. This is what puzzled me.

Perhaps you know a solution.

andimik, 2017-03-21:
Do you use 32 or 64 bit? I fear that the online installer of qt only loads qt 5.8 in 64 bit environment.

Wikinaut, 2017-03-21:
64bit. I will try to deinstall and re-install the whole beast. I also hate them to have only a binary (`...run`) installer.

Wikinaut, 2017-03-21:
by the way, I could build https://github.com/JvanKatwijk/qt-dab and it works with my https://www.amazon.de/gp/product/B00VZ1AWQA

Wikinaut, 2017-03-21:
Okay, ich bin weiter....
* deleted Makefile and welle.io.pro.user files
* restarted QTCreator, now I saw two selections "Desktop" and "Qt5.8 64bit Desktop..." which was unselected. After selection this one, build now says

"libairspy.h" no such file

perhaps because I did not yet installed them. I will report here later.

andimik, 2017-03-21:
You should follow the prerequisites. When you uncomment Airspy you have to have the Airspy files installed of course.

Wikinaut, 2017-03-21:
I guess, you meant "when you do not uncomment Airspy you have to have ... installed".

On my Linux Mint 18.1 (= Ubuntu 16.04), I had to also install `libpulse-dev` - see my pr.

Off topic:

welle.io works, but when compared to qt-dab, there are many reception problems (FIC CRC errors) with welle.io (when using the same hardware). It's a little better when I use high manual gain.