AlbrechtL/welle.io #27: AGC problems

Wikinaut, 2017-03-21:
I can only get welle.io working here with manual gain (12,50 or so), which results in SNR 14dB.
AGC results in unreliable rx quality, and SNR decreases !

AlbrechtL, 2017-03-21:
This problem appears of the signal strength is to strong at your location. Does this also happens with qt-dab?

Wikinaut, 2017-03-22:
No. qt-dab works very well, this is why I suppose it may be a software problem. On some stations, I see a very strong pilot at the mid frequency (Träger), on others not.

By the way, why don't you use "dB" at the vertical axis ?

Wikinaut, 2017-03-23:
Pls. let me know, if you need more details, or if I can help with further tests.

Wikinaut, 2017-03-23:
@AlbrechtL what are reasonable good SNR values ?

andimik, 2017-03-23:
There is no general rule for that. You have to try it. Furthermore, every stick supports different gain values.

The biterror rate is a factor, too.

But as this is obviously a mirror of 12C (you receive no other transmission) it really makes no sense to continue on 44 MHz.

Wikinaut, 2017-03-23:
As already mentioned, qt-dab works. Perhaps you have an idea, where I can start investigating, I still suppose it's a code problem with your code.

AlbrechtL, 2017-03-23:
Do you mean that AGC is working well with qt-dab?

Wikinaut, 2017-03-23:
Yes. I am just checking that. DRadio signal is a strong one (I will check another soon):

* autogain is on

[image]

Wikinaut, 2017-03-23:
Here's another one ("RBB Antenne" on channel 7D)

[image]

andimik, 2017-03-23:
Interesting for me: Different Muxes need different CPU power.

Wikinaut, 2017-03-23:
Wait: I just toggled qt-dab setting for autogain. It *might* be that the displayed setting is *not* the correct one!!!

Wikinaut, 2017-03-23:
Ah interesting! I have to report this to qt-dab as well: when I play with the agc setting, the reception gets worse. Only when I rescan the band (here: 7D) it works again good. 

May this be the reason ? Is there a difference when toggling agc ?
* on a station
* per band (pilot) ?

Wikinaut, 2017-03-23:
(afk for 45 minutes)

Wikinaut, 2017-03-23:
@andimik the two have different protection levels.

AlbrechtL, 2017-03-24:
I added an experimental new AGC for rtl-sdr. This AGC is implemented in welle.io instead of using the original rtl-sdr AGC.

Please try it again with the version https://github.com/AlbrechtL/welle.io/commit/01c2f7662f855c9aa8117d8daf16b7c2be83f93e and report your results.

Wikinaut, 2017-03-24:
:1 :1 Perfect!

Wikinaut, 2017-03-24:
SNR values here around 19 dB. Quicker feedback was not possible :-)

Wikinaut, 2017-03-24:
How can I send you a donation ?

Wikinaut, 2017-03-25:
It looks as if stronger stations (when using AGC) are still causing problems. I cannot see frame sync, RS, or frame errors in the expert window, but the spectrum looks different than others, and audio signal is muted now and then (drop out?  muted ?)

When not using AGC and using manual gain, these stations can be played without drop outs. (Example: manual gain 14, SNR 7..9 dB works; switching to AGC: drop outs)

AlbrechtL, 2017-03-25:
I did some improvements. Please try it again and report your results.

Wikinaut, 2017-03-25:
:+1: quick feedback: no drop-outs any more. Also the spectrum view appears to look better.

Wikinaut, 2017-03-25:
Correction: now the reception has many drop outs. So I now suppose it may be a temperature problem with the stick. Will try to reproduce the issue (switching off, cooling, retrying), perhaps returning the stick.

Wikinaut, 2017-03-25:
Again: qt-dab works well. Strange.....

Wikinaut, 2017-03-25:
In welle.io, what does frequency correction mean ? After a restart of it, welle.io works.

AlbrechtL, 2017-03-25:
The frequency correction is part of the synchronization to the channel. If you use a cheap stick without temperature compensation the frequency is shifting permanently. So it needs to corrected. This part of the code is buggy.  Sometimes welle.io is running out of sync and needs a resynchronization. But qt-dab should have the same problem because welle.io uses qt-dab code.
For a short workaround it also helps just to switch to another frequency.

Wikinaut, 2017-03-25:
@AlbrechtL I tested qt-dab with Radio GOLD more than 3 hours: no drop-outs.

But with welle.io, unfortunately, there are still issues (drop-outs). Also the spectrum looks strange. Let me know, if you want me to post some screen shots.

AlbrechtL, 2017-03-26:
Do you have the sound issues also with disabled expert view?

AlbrechtL, 2017-03-26:
I opened a new issue https://github.com/AlbrechtL/welle.io/issues/35 for the expert view sound problem.

Wikinaut, 2017-03-26:
@AlbrechtL 
> Do you have the sound issues also with disabled expert view?
Tests started. Quick response: you seem to be right, may be now an expert-view-causes-sound-drops problem. 

I will report in #35, if I notice further drops. I am just enjoying the Sonntagskonzert on RBB kulturradio. Shown gains (in the terminal window) are currently between 19,7 and 22,9 and once I see 25,4 dB.

Wikinaut, 2017-03-26:
BTW: I always started welle.io from within the QTCreator on a 8GB Linux Mint 18.1 (=Ubuntu 16.04).

Wikinaut, 2017-03-26:
I think, this can be closed. I leave it up to you.

AlbrechtL, 2017-03-26:
I think so too.

closed 2017-03-26 by AlbrechtL