AlbrechtL/welle.io #29: Uknown option B

tnadolecki, 2017-03-22:
I found another signal that looks like DAB but it's at around 44 Mhz, Can't search for it so I decided to do it via rawfile. Unfortunately I can't do it the way it is described in manual.
Any help would be apreciated.
It's latest windows10 version.

C:\Program Files (x86)\welle.io>welle-io.exe -D rawfile -F test.raw -B s16le
Unknown option 'B'.

Thanks. Tom.

tnadolecki, 2017-03-22:
I guess the manual is old ?

andimik, 2017-03-22:
You are using Windows. Please wait for a new release.

andimik, 2017-03-22:
44 MHz? This has nothing to do with DAB

Although it is specified for VHF I, even 44 MHz would be too low. The lowest ever used active Frequency ist channel 5A (174 MHz)

Maybe you have received a mirror signal?

tnadolecki, 2017-03-22:
Yes, it was definitely mirror and it was showing up only in windows. Now I'm back on Linux and it's gone. Must be something to do with drivers. 
I'm done with windows anyways. Thanks.

closed 2017-03-22 by tnadolecki

tnadolecki, 2017-03-22:
Actually there is signal at 44.175 Mhz I'm confused as what it is. 
Here is RAW GQRX file, 20 seconds recording if you don't mind checking.
http://matkaboska.sytes.net/s/HI2fT2wgLL0hqOl
It does look exactly like DAB at 217.360 here.

andimik, 2017-03-22:
217,360 MHz is not a DAB frequency. You mean 227,360 (12C) or 213,360 (10C)

I've downloaded the file and will test it.

andimik, 2017-03-22:
Where are you living and what Mux should we see here? I suppose you are from Ireland:  http://www.fmlist.org/sendertabelle/dab-ww_more.php?inputLand=IRL&inputEn=2E01

Please keep in mind that raw is not raw.

I wanted to convert it to a readable format for welle.io, but I got no sync (both in qt-dab nor welle.io). You need sow for that: `sudo apt install sox`

`sox -t raw -e floating-point -b 32 -c 2 -r 2048000 gqrx_20170322_125234_44171230_2048000_fc.raw -t raw -e signed-integer -c 2 -b 16 -r 2048000 -L gqrx_20170322_125234_44171230_2048000_s16le.raw`

Are you sure the center frequency is correct?

AlbrechtL, 2017-03-22:
I'm not familiar with the GQRX file format. In which format is it?

@Andimik: Do you see an OFDM spectrum with your converted file.

andimik, 2017-03-22:
No, not at all.

The gqrx spec I found on http://gnuradio.4.n7.nabble.com/GQRX-raw-IQ-file-format-structure-td55448.html

tnadolecki, 2017-03-23:
Sorry for late response. Just to clarify - I'm in Limerick, Ireland, receiving 227.360 (12C) from Woodcock Hill  trasmitter. It's an RTE mux. That is working fine except for DAB (which is majority of programme lol)
The other signal I recorded is centered at exactly 44.175 Mhz. Looks just like the signal I'm receiving on 12C, same bandwidth and all...
Would you like me to record another IQ file in different format ? 
PS: Great stuff BTW, I was long waiting for this s/w. Thumbs up all the way :)
Thanks,
Tom.

tnadolecki, 2017-03-23:
Hi, I uploaded another raw iq file of that unknown 44.175 Mhz signal. I tried to play in welle.io , it picks up sync and shows very low SNR however can't make much of it.
http://matkaboska.sytes.net/s/ALJ9ErTe7DrKfub
Thanks. Tom.

andimik, 2017-03-23:
You forgot -g which is the gain switch 

Only Sticks with E4000 have an automatic gain.

andimik, 2017-03-23:
Thanks for the file. If you rename it to raw then you can use it in qt-dab - or better - in eti-stuff.

It is DAB Ireland Mux1.

So don't continue on 44 MHz, as you have it on 12C with a better signal, I suppose.

tnadolecki, 2017-03-23:
They must be testing as the frequency has just shifted down few hundred khz. I'm not bothered since it's not usable level signal.
Thanks for finding it out.
Tom.

andimik, 2017-03-23:
Anyway, you found much more out that you might have thought. See new issue on Qt-DAB :-)

tnadolecki, 2017-03-23:
Not a problem, I do that all the time. I'm here to help if you need me again lol :)
But seriously I'm constructing a small magnetic loop antenna to catch some more of DAB signal if you still need it. 
Thanks.
Tom.

andimik, 2017-03-23:
Try 12A, which is the 2nd mux in Ireland, but Limerick is not covered at the moment, so good luck!

http://fmscan.org/net.php?r=d&m=m&itu=IRL&bouq=1&pxf=DAB+Ireland+Mux2