AlbrechtL/welle.io #30: Breaking audio and warnings in linux

tnadolecki, 2017-03-22:
Hi, After having probs with Windows version I gave it a try on Linux.
Audio is horribly broken and I get this repetitive message printed in thousands of lines.

fib-processor: we found for "RTE 2FM +       " serviceId 8802
DABConstants: Unknown language type
dab-audio: we have now DAB+
Warning: No standard extension payload allowed in DRM
Audio: Sample rate 48000 Hz
Audio: ActiveState
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM

AlbrechtL, 2017-03-22:
Which Linux do you are using?
The message "Warning: No standard extension payload allowed in DRM" comes from faad.
Maybe this helps: https://forums.gentoo.org/viewtopic-t-912218-start-0.html

tnadolecki, 2017-03-23:
This appears to be fixed with latest 0dd9bcb
I'm using Linux dell 4.4.0-66-generic #87-Ubuntu SMP 
I did not upgrade faad.
Thanks.

AlbrechtL, 2017-03-23:
I think you can into this problem https://github.com/AlbrechtL/welle.io/issues/14 because it didn't change anything regarding to faad.

tnadolecki, 2017-03-23:
I have just seen it again after long use of program. It might be just my own machine then. I apologize for confusion. I will keep an eye on it and advise of any findings.
BTW the latest version seems to have fix all my problems.
Thank you again.

tnadolecki, 2017-03-23:
dpkg -l | grep faad
ii  gstreamer1.0-plugins-bad-faad:amd64           1.8.3-1ubuntu0.2                                    amd64        GStreamer faad plugin from the "bad" set
ii  libfaad-dev:amd64                             2.8.0~cvs20150510-1                                 amd64        freeware Advanced Audio Decoder - development files
ii  libfaad2:amd64                                2.8.0~cvs20150510-1                                 amd64        freeware Advanced Audio Decoder - runtime files

I still get those errors sporadically.

bolero@dell:~/build-welle.io-Desktop_Qt_5_8_0_GCC_64bit-Debug$ ./welle-io 
QML debugging is enabled. Only use this in a safe environment.
InputFactory: Input device: "auto"
Airspy: Open airspy
Airspy: airpsy_open () failed: AIRSPY_ERROR_NOT_FOUND ( -5 )
RTL_TCP: Open rtl-sdr
RTL_TCP: Found 1 devices. Uses the first one
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
RTL_TCP: Supported gain values 29
RTL_TCP: gain 49.6
RTL_TCP: gain 48
RTL_TCP: gain 44.5
RTL_TCP: gain 43.9
RTL_TCP: gain 43.4
RTL_TCP: gain 42.1
RTL_TCP: gain 40.2
RTL_TCP: gain 38.6
RTL_TCP: gain 37.2
RTL_TCP: gain 36.4
RTL_TCP: gain 33.8
RTL_TCP: gain 32.8
RTL_TCP: gain 29.7
RTL_TCP: gain 28
RTL_TCP: gain 25.4
RTL_TCP: gain 22.9
RTL_TCP: gain 20.7
RTL_TCP: gain 19.7
RTL_TCP: gain 16.6
RTL_TCP: gain 15.7
RTL_TCP: gain 14.4
RTL_TCP: gain 12.5
RTL_TCP: gain 8.7
RTL_TCP: gain 7.7
RTL_TCP: gain 3.7
RTL_TCP: gain 2.7
RTL_TCP: gain 1.4
RTL_TCP: gain 0.9
RTL_TCP: gain 0
Audio: ActiveState
GUI: AGC on
qml: dpiScaleFactor: 1
GUI: AGC off
GUI: Starting true
OFDM-processor: start
OFDM-processor: closing down
OFDM-processor: start
Audio: StoppedState
OFDM-processor: closing down
OFDM-processor: start
GUI: "12C" -> 227360000
fib-processor: we found for "RTE 2FM         " serviceId 8706
DABConstants: Unknown program type
DABConstants: Unknown language type
dab-audio: we have now DAB
Audio: Sample rate 48000 Hz
Audio: ActiveState
fib-processor: we found for "RTE 2FM +       " serviceId 8802
DABConstants: Unknown language type
dab-audio: we have now DAB+
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
fib-processor: we found for "RTE 2XM         " serviceId 8773
DABConstants: Unknown program type
DABConstants: Unknown language type
dab-audio: we have now DAB
fib-processor: we found for "RTE 2XM         " serviceId 8773
DABConstants: Unknown program type
DABConstants: Unknown language type
dab-audio: we have now DAB
fib-processor: we found for "RTE GOLD        " serviceId 8776
DABConstants: Unknown language type
dab-audio: we have now DAB
fib-processor: we found for "RTE Jr Radio    " serviceId 8775
DABConstants: Unknown language type
dab-audio: we have now DAB
fib-processor: we found for "RTE LYRIC       " serviceId 8707
DABConstants: Unknown language type
dab-audio: we have now DAB
fib-processor: we found for "RTE PULSE       " serviceId 8777
DABConstants: Unknown language type
dab-audio: we have now DAB

andimik, 2017-03-23:
If you don't have an Airspy dongle, you could comment out the respective line in the pro file.

AlbrechtL, 2017-03-23:
Do you still have the audio problems? What happens if you get the line `Warning: No standard extension payload allowed in DRM`?

tnadolecki, 2017-03-23:
Audio breaks, namely goes silent for split of second, then split of second of audio is on again etc.. kind of stuttering sound.

tnadolecki, 2017-03-23:
Update: It's clearly happening only on + channels. (I can't believe I have not noticed it previously)

fib-processor: we found for "RTE Radio 1Extra" serviceId 8778
DABConstants: Unknown language type
dab-audio: we have now DAB
fib-processor: we found for "RTE Radio 1 +   " serviceId 8801
DABConstants: Unknown language type
dab-audio: we have now DAB+
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
fib-processor: we found for "RTE Radio 1     " serviceId 8705
DABConstants: Unknown language type
dab-audio: we have now DAB
fib-processor: we found for "RTE Radio 1 +   " serviceId 8801
DABConstants: Unknown language type
dab-audio: we have now DAB+
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM
fib-processor: we found for "RTE Radio 1     " serviceId 8705
DABConstants: Unknown language type
dab-audio: we have now DAB
fib-processor: we found for "RTE 2FM +       " serviceId 8802
DABConstants: Unknown language type
dab-audio: we have now DAB+
Warning: No standard extension payload allowed in DRM
Warning: No standard extension payload allowed in DRM

AlbrechtL, 2017-03-23:
Is it possible that you send me a [RAW file](https://www.welle.io/devices/rawfile#record-an-u8-raw-file) which shows the problem?

tnadolecki, 2017-03-23:
It does when I play this file 
http://matkaboska.sytes.net/s/tvascXIUIsC8UBe

tnadolecki, 2017-03-24:
Please see attached video demontrating the problem.
https://youtu.be/HucywdpP2Ak

AlbrechtL, 2017-03-24:
I tested your file. On my Ubuntu 16.04 LTS system it is running without any problems. Are you sure that you are using the original FAAD library from Ubuntu?

andimik, 2017-03-24:
I have no problems with the sound as well.

> $ ./welle-io -D rawfile -F 227360000-short.iq.raw 
> QML debugging is enabled. Only use this in a safe environment.
> InputFactory: Input device: "rawfile"
> RAWFile Period = 8000
> Audio: ActiveState
> GUI: AGC on
> qml: dpiScaleFactor: 1
> GUI: AGC off
> GUI: Starting true
> OFDM-processor: start
> OFDM-processor: closing down
> OFDM-processor: start
> Audio: StoppedState
> OFDM-processor: closing down
> OFDM-processor: start
> GUI: "5A" -> 174928000
> fib-processor: we found for "RTE 2FM         " serviceId 8706
> DABConstants: Unknown program type
> DABConstants: Unknown language type
> dab-audio: we have now DAB
> Audio: Sample rate 48000 Hz
> Audio: ActiveState
> RAWFile: End of file, restarting
> fib-processor: we found for "RTE GOLD        " serviceId 8776
> DABConstants: Unknown program type
> DABConstants: Unknown language type
> dab-audio: we have now DAB

But I found out that the slideshows are not displayed, they even miss in welle.io ...

tnadolecki, 2017-03-24:
dpkg -l faad
ii  faad           2.8.0~cvs201 amd64        freeware Advanced Audio Decoder p

This is what I have and I believe it is original from ubuntu. I have removed all external /etc/apt/sources.list.d/ entries also rechecked /etc/apt/sources.list for any external links. All seems to be original now.
(And of course I did apt-get update followed by upgrade.)
Since I'm the only one with the problem I think it is OK to close the issue.
Regards.
T.

AlbrechtL, 2017-03-24:
OK. We can open this issue again if other persons are running into the same problem.

andimik, 2017-03-26:
BTW: I have the same faad installed you mentioned. 

My Ubuntu is 16.04 LTS.

ROV28, 2018-10-13:
I run into the same problem: "FaadDecoder: warning: No standard extension payload allowed in DRM" and finally Welle.io crashes (Segmentation fault)

Running Welle.io on Fedora 28, faad version: faad2-1:2.8.8-1.fc28.x86_64 : Open source MPEG-4 and MPEG-2 AAC decoder

AlbrechtL, 2018-10-13:
Did you compile libfaad yourself?

ROV28, 2018-10-14:
No I didn't, the faad lib is provided by the  fedora-multimedia repository (faad2-2.8.8-1.fc28.src.rpm).

AlbrechtL, 2018-11-10:
Solved.
The DRM support in libfaad has to be disabled (see https://github.com/AlbrechtL/welle.io/issues/314).

ROV28, 2018-11-10:
I do not want to nag, but disabling DRM support in libfaad means that you cannot use the libfaad from the repositories and you might also kill other applications using libfaad with DRM support. 

Personally I do not think this is the right way to go, I would rather see that Welle.io would be compatible with the newer versions of libfaad.

andimik, 2019-01-04:
I have now the same problem. I wanted to test DRMplus and installed some DRM-enabled packages but now none of the DAB programs (dablin, qt-dab, etc.) will play DAB+ and have lots of these

`Warning: No standard extension payload allowed in DRM`

AlbrechtL, 2019-01-07:
I think this is a libfaad bug.

mpbraendli, 2019-01-08:
I did a small experiment, on Arch Linux, where the faad2 package is not compiled with `--with-drm`. I modified the package, enabled DRM and welle.io always crashes.

Looking at the faad2 source code, in `libfaad/common.h` line 137:
```
#ifdef DRM
# ifndef ALLOW_SMALL_FRAMELENGTH
#  define ALLOW_SMALL_FRAMELENGTH
# endif
# undef LD_DEC
# undef LTP_DEC
# undef MAIN_DEC
# undef SSR_DEC
#endif
```
This disables `MAIN_DEC`, which makes the function `can_decode_ot()` fail, which makes `NeAACDecSetConfiguration()` fail. I don't know if we're using it wrongly in our `AACDecoderFAAD2` or if something is broken with DRM, as there is no explanation why DRM decoding is not enabled by default in faad2's `configure.ac`.

I won't pursue this research, but it can give some hints for somebody who wants to look into that.

Aside: Why is there faad code in https://github.com/AlbrechtL/welle.io/tree/next/src/libs/faad2/libfaad ?

basicmaster, 2019-01-13:
Hmm, I'm really curious why v2.8.8 makes problems; the version from audiocoding.com compiles both the DRM and the non-DRM lib version. Also Ubuntu 18.04 ships both versions....

mpbraendli, 2019-01-15:
ubuntu ships two `.so` files: https://packages.ubuntu.com/bionic/amd64/libfaad2/filelist
as does fedora 29: https://rpmfind.net/linux/RPM/rpmfusion/free/fedora/29/x86_64/Packages/f/faad2-libs-2.8.8-2.fc29.i686.html
Why does the DRM stuff pose an issue if the non-`_drm.so` has not been compiled with DRM enabled?
Does welle-io link against `libfaad_drm.so` ?

AlbrechtL, 2019-01-16:
I expect that the linker uses `libfaad.so`.

Inside the Travic CI logs you can see that the non-DRM version is used
[image]

AlbrechtL, 2024-10-08:
I'm closing this issue because I'm not sure if it is still relevant. If it still occurs in the latest version, we can reopen it.

closed 2024-10-08 by AlbrechtL