AlbrechtL/welle.io #31: Improve rawfile support

andimik, 2017-03-23:
A user does normally not care about rawfile formats (BTW: welle-io was the first page to let me know the difference between *.sdr and *.raw file structure :+1 )

So it would be good to implement an autodetection between u8 and s16le.

Furthermore, welle-io's behavior in scanning is the same as for VHF, after pressing "Channel scan Start" 5A will be analyzed, but then suddenly the other channels (5B, etc.) follow. I would expect that only one channel is used.

andimik, 2017-04-04:
BTW: I've asked ODR developers to include u8 as export format ...

AlbrechtL, 2017-04-11:
In 3392297 I added the following:
 * Signed 8 bit format (s8) for ODR RAW files
 * Disabled channel scan for RAW files

An automatic detection of the format is not possible because this information is missing inside the RAW files. The only option is to try formats and look if we get a sync. But this can also the user do. BTW: The RAW file input is for developers and those should know which format is used. Users can also read the [rawfile documentation](https://www.welle.io/devices/rawfile).

closed 2017-04-11 by AlbrechtL

andimik, 2017-04-11:
I'm happy about s8 ;-) Yes it works great!

Format detection was only before I learned the difference of raw file formats as I had absolutely no idea before