AlbrechtL/welle.io #32: AppImage support

mantielero, 2017-03-23:
Distributing the application as an [AppImage](http://appimage.org/) can make easier the installation/execution under much more Linux Distributions.

AlbrechtL, 2017-03-24:
I added it to the [task list](https://github.com/AlbrechtL/welle.io/wiki/Open-Tasks). However, somebody has to take the job. Are you interested in creating one?

mantielero, 2017-03-24:
I've been trying but I'm not managing to do so. (First time I try to build an AppImage and I don't know much about qtcreator/qt programming).
I have manage to compile welle and run it. But when I run:
```
$ linuxdeployqt-continuous-x86_64.AppImage ./welle-io
Not using FHS-like mode
app-binary: "/tmp/build-welle.io-Desktop-Release/welle-io"
appDirPath: "/tmp/build-welle.io-Desktop-Release"
relativeBinPath: "welle-io"
ERROR: Bundling Qt provided by distributions is not yet supported by this tool.
ERROR: Please see https://github.com/probonopd/linuxdeployqt/issues/79 and
ERROR: submit a Pull Request to implement this.
ERROR: In the meantime, use Qt from the Qt Company
ERROR: or Qt for /opt from https://launchpad.net/~beineri
```
I don't know how to properly setup Qt for this to work.

Besides I have checked that I don't get any DAB+ station where I live, so I have lost a lot of motivation. 

Sorry

tnadolecki, 2017-03-24:
Would a RAW file motivate you ?

probonopd, 2017-03-25:
> I don't know how to properly setup Qt for this to work.

On Ubuntu 14.04 (e.g., on [Travis CI](https://travis-ci.org/)), run:

```
sudo add-apt-repository ppa:beineri/opt-qt58-trusty -y
sudo apt-get update -qq
sudo apt-get -y install qt58base
source /opt/qt58/bin/qt58-env.sh
```

Here is a full example for an other Qt-based application:

https://github.com/jeena/FeedTheMonkey/blob/master/.travis.yml

probonopd, 2017-03-25:
I am trying to build an AppImage on Ubuntu 14.04 on Travis CI but I am running into

```
g++ -c -pipe -O2 -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DHAVE_AIRSPY -DHAVE_RTLSDR -DQT_NO_DEBUG -DQT_QUICK_LIB -DQT_CHARTS_LIB -DQT_WIDGETS_LIB -DQT_MULTIMEDIA_LIB -DQT_GUI_LIB -DQT_QML_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -D_REENTRANT -I. -I. -I. -Isrc -Isrc/ofdm -Isrc/backend -Isrc/backend/audio -Isrc/backend/data -I./src/backend/data/journaline -Isrc/output -Isrc/various -Isrc/input -Isrc/gui -isystem /usr/local/include -I/opt/qt58/include -I/opt/qt58/include/QtQuick -I/opt/qt58/include/QtCharts -I/opt/qt58/include/QtWidgets -I/opt/qt58/include/QtMultimedia -I/opt/qt58/include/QtGui -I/opt/qt58/include/QtQml -I/opt/qt58/include/QtNetwork -I/opt/qt58/include/QtCore -I. -isystem /usr/include/libdrm -isystem /usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/opt/qt58/mkspecs/linux-g++ -o pad_decoder.o src/backend/data/pad_decoder.cpp
src/backend/audio/dab-audio.cpp: In member function ‘virtual void dabAudio::run()’:
src/backend/audio/dab-audio.cpp:130:33: error: variable-sized object ‘Data’ may not be initialized
 int16_t Data [fragmentSize] = {0};
                                 ^
src/backend/audio/dab-audio.cpp:131:34: error: variable-sized object ‘tempX’ may not be initialized
 int16_t tempX [fragmentSize] = {0};
                                  ^
make: *** [dab-audio.o] Error 1
make: *** Waiting for unfinished jobs....
The command "make -j4" exited with 2.
```

probonopd, 2017-03-25:
Here is my preliminary `.travis.yml` but it still suffers from the above:
https://github.com/probonopd/welle.io/blob/patch-1/.travis.yml

probonopd, 2017-03-25:
@AlbrechtL can you give me step-by-step instructions how to compile this on Ubuntu 14.04 please? Then I can do an AppImage.

AlbrechtL, 2017-03-25:
Please use the [Ubuntu 16.04 how to](https://github.com/AlbrechtL/welle.io#ubuntu-linux-1604-lts) and adapt it to 14.04.

probonopd, 2017-03-25:
That's what I did, [here](https://github.com/probonopd/welle.io/blob/patch-1/.travis.yml).

AlbrechtL, 2017-03-25:
Can you provide a welle.io AppImage to test it?

probonopd, 2017-03-25:
No, because compilation fails with [this](https://github.com/AlbrechtL/welle.io/issues/32#issuecomment-289198249) error.

AlbrechtL, 2017-03-27:
If you change 
```
 int16_t Data [fragmentSize] = {0};
 int16_t tempX [fragmentSize] = {0};
```
to
```
 int16_t Data [fragmentSize] ;
 int16_t tempX [fragmentSize];
```
it should work.

probonopd, 2017-03-27:
Thanks, how can I disable `airspy` at build time without editing the .pro file?

AlbrechtL, 2017-03-28:
I don't know. You can check the qmake docu or you write a small script that edits the .pro file before the build starts.
Another option is to include the airspy driver to the image.

probonopd, 2017-03-28:
Doing that now. Thanks

probonopd, 2017-03-28:
The command `sudo make INSTALL_ROOT=appdir install` fails, do you have an idea why; or what I could do?

probonopd, 2017-03-28:
Turns out `make install` does not work at all?!

probonopd, 2017-03-31:
I think this can be closed now, as an AppImage gets built on each `git push` now. The download link is in the Travis CI build log.

Something like https://github.com/probonopd/uploadtool could be used to upload the builds to GitHub Releases.

AlbrechtL, 2017-03-31:
Thank you very much for the AppImage support!

closed 2017-03-31 by AlbrechtL