AlbrechtL/welle.io #35: Sound drops-outs with enabled expert view

AlbrechtL, 2017-03-26:
With enabled expert view the sound output failed sometimes. It seems that something goes wrong with the threads (all threads are calculated on one core).
Observed on Fedora 25.

Wikinaut, 2017-03-26:
@AlbrechtL since having closed the expert view this morning, and even after having changed stations, there were no drop-outs any more.

AlbrechtL, 2017-03-26:
It seems that the qtcharts module blocks the app for some milliseconds.

zygmund2000, 2017-03-28:
Yes, without fft view sound is ok.

AlbrechtL, 2017-03-28:
It is also discovered with Ubuntu 16.06 LTS, Manjaro Linux, Windows 7 and 10 and Mac OS X.

AlbrechtL, 2017-03-28:
It's a bug inside QT 5.8 https://bugreports.qt.io/browse/QTBUG-59040

kgarrels, 2017-03-30:
It did a build with Qt5.7. Seems to work in expoert mode as well.
Some different messages on the console though:
```

Starting /Users/kai/gnuradio/welle.io/welle-io.app/Contents/MacOS/welle-io...
QML debugging is enabled. Only use this in a safe environment.
InputFactory: Input device: "auto"
Airspy: Open airspy
Airspy: 2 sample rates are supported
Airspy: sample rates: 0 10000000
Airspy: sample rates: 1 2500000
Airspy: selected samplerate 2500000
Audio: ActiveState
GUI: AGC on
qml: dpiScaleFactor: 1
GUI: AGC off
Audio: IdleState
Audio: StoppedState
Button_QMLTYPE_56 QVariant(Invalid) QRect(787,928 -40x-40)
Button_QMLTYPE_56 QVariant(Invalid) QRect(787,928 -40x-40)
Slider_QMLTYPE_78 QVariant(Invalid) QRect(799,756 0x46)
Slider_QMLTYPE_78 QVariant(Invalid) QRect(799,756 0x46)
TouchProgressBar_QMLTYPE_77 QVariant(Invalid) QRect(799,615 0x60)
TouchProgressBar_QMLTYPE_77 QVariant(Invalid) QRect(799,615 0x60)
GUI: Starting true
OFDM-processor: start
OFDM-processor: closing down
OFDM-processor: start
OFDM-processor: closing down
OFDM-processor: start
GUI: "9D" -> 208064000
fib-processor: we found for "DASDING         " serviceId 54181
DABConstants: Unknown language type
dab-audio: we have now DAB+
Audio: Sample rate 48000 Hz
Audio: ActiveState
Button_QMLTYPE_56 QVariant(Invalid) QRect(787,928 -40x-40)
Button_QMLTYPE_56 QVariant(Invalid) QRect(787,928 -40x-40)
Slider_QMLTYPE_78 QVariant(Invalid) QRect(799,756 0x46)
Slider_QMLTYPE_78 QVariant(Invalid) QRect(799,756 0x46)
TouchProgressBar_QMLTYPE_77 QVariant(Invalid) QRect(799,615 0x60)
TouchProgressBar_QMLTYPE_77 QVariant(Invalid) QRect(799,615 0x60)
OFDM-processor: closing down
OFDM-processor: start
Audio: StoppedState
OFDM-processor: closing down
OFDM-processor: start
GUI: "5C" -> 178352000
fib-processor: we found for "ERF Plus        " serviceId 6756
DABConstants: Unknown language type
dab-audio: we have now DAB+
Button_QMLTYPE_56 QVariant(Invalid) QRect(787,928 -40x-40)
Button_QMLTYPE_56 QVariant(Invalid) QRect(787,928 -40x-40)
Slider_QMLTYPE_78 QVariant(Invalid) QRect(799,756 0x46)
Slider_QMLTYPE_78 QVariant(Invalid) QRect(799,756 0x46)
TouchProgressBar_QMLTYPE_77 QVariant(Invalid) QRect(799,615 0x60)
TouchProgressBar_QMLTYPE_77 QVariant(Invalid) QRect(799,615 0x60)
OFDM-processor: closing down
OFDM-processor: start
OFDM-processor: closing down
OFDM-processor: start
GUI: "11B" -> 218640000
fib-processor: we found for "HITRADIO OHR    " serviceId 54285
dab-audio: we have now DAB+
Audio: ActiveState
``

AlbrechtL, 2017-03-30:
> Button_QMLTYPE_56 QVariant(Invalid) QRect(787,928 -40x-40)
> Button_QMLTYPE_56 QVariant(Invalid) QRect(787,928 -40x-40)
> Slider_QMLTYPE_78 QVariant(Invalid) QRect(799,756 0x46)
> Slider_QMLTYPE_78 [...]

Interesting. Is everything working inside the GUI?
It seems that switching back to QT 5.7 is not easily possible without any work.

kgarrels, 2017-03-30:
yes, everything works quite normal, no special observations so far.

AlbrechtL, 2017-03-30:
Funny, I can't get your error messages with with QT 5.7.1

kgarrels, 2017-03-30:
one thing does not work: the program crashes on quit in thread 9:
```
Process:               welle-io [5194]
Path:                  /Users/USER/*/welle-io.app/Contents/MacOS/welle-io
Identifier:            at.welle-io
Version:               0
Code Type:             X86-64 (Native)
Parent Process:        Qt Creator [3904]
Responsible:           welle-io [5194]
User ID:               501

Date/Time:             2017-03-30 22:42:21.171 +0200
OS Version:            Mac OS X 10.12.4 (16E195)
Report Version:        12
Anonymous UUID:        1AC852BA-1FE1-716E-BDD7-2A219EE16E13

Sleep/Wake UUID:       FF4BEA9C-AE13-4E3A-9017-84D74815DA67

Time Awake Since Boot: 86000 seconds

System Integrity Protection: enabled

Crashed Thread:        9  com.apple.audio.IOThread.client

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x00000001179d0120
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [0]

VM Regions Near 0x1179d0120:
    MALLOC_LARGE           0000000116f83000-00000001179c3000 [ 10.2M] rw-/rwx SM=PRV  
--> 
    __TEXT                 00000001179db000-00000001179e0000 [   20K] r-x/rwx SM=COW  /System/Library/Extensions/AppleHDA.kext/Contents/PlugIns/AppleHDAHALPlugIn.bundle/Contents/MacOS/AppleHDAHALPlugIn

Thread 0:: Dispatch queue: com.apple.main-thread
0   libsystem_malloc.dylib        	0x00007fffb2c88dd9 free + 4
1   org.qt-project.QtCore         	0x000000010f3ef2b9 QHash<QString, QString>::deleteNode2(QHashData::Node*) + 57
2   org.qt-project.QtCore         	0x000000010f204be9 QHashData::free_helper(void (*)(QHashData::Node*)) + 73
3   org.qt-project.QtCore         	0x000000010f3ef473 QMimeTypePrivate::~QMimeTypePrivate() + 307
4   org.qt-project.QtCore         	0x000000010f3ed6ac QMimeType::~QMimeType() + 44
5   org.qt-project.QtCore         	0x000000010f3fccc2 QHash<QString, QMimeType>::deleteNode2(QHashData::Node*) + 18
6   org.qt-project.QtCore         	0x000000010f204be9 QHashData::free_helper(void (*)(QHashData::Node*)) + 73
7   org.qt-project.QtCore         	0x000000010f3fcb54 QMimeXMLProvider::~QMimeXMLProvider() + 452
8   org.qt-project.QtCore         	0x000000010f3fc7de QMimeXMLProvider::~QMimeXMLProvider() + 14
9   org.qt-project.QtCore         	0x000000010f3eca80 (anonymous namespace)::Q_QGS_staticQMimeDatabase::innerFunction()::Cleanup::~Cleanup() + 32
10  libsystem_c.dylib             	0x00007fffb2b9a178 __cxa_finalize_ranges + 332
11  libsystem_c.dylib             	0x00007fffb2b9a4b2 exit + 55
12  libdyld.dylib                 	0x00007fffb2b0523c start + 8

Thread 1:: org.libusb.device-hotplug
0   libsystem_kernel.dylib        	0x00007fffb2c2c34a mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fffb2c2b797 mach_msg + 55
2   com.apple.CoreFoundation      	0x00007fff9d3a34e4 __CFRunLoopServiceMachPort + 212
3   com.apple.CoreFoundation      	0x00007fff9d3a2971 __CFRunLoopRun + 1361
4   com.apple.CoreFoundation      	0x00007fff9d3a21c4 CFRunLoopRunSpecific + 420
5   com.apple.CoreFoundation      	0x00007fff9d3e1c31 CFRunLoopRun + 97
6   libusb-1.0.0.dylib            	0x000000010e858249 darwin_event_thread_main + 411
7   libsystem_pthread.dylib       	0x00007fffb2d1e9af _pthread_body + 180
8   libsystem_pthread.dylib       	0x00007fffb2d1e8fb _pthread_start + 286
9   libsystem_pthread.dylib       	0x00007fffb2d1e101 thread_start + 13

Thread 2:: ofdmDecoder
0   libsystem_kernel.dylib        	0x00007fffb2c33bf2 __psynch_cvwait + 10
1   libsystem_pthread.dylib       	0x00007fffb2d1f86e _pthread_cond_wait + 712
2   org.qt-project.QtCore         	0x000000010f1aa0ef QWaitConditionPrivate::wait_relative(unsigned long) + 287
3   org.qt-project.QtCore         	0x000000010f1a9f2b QWaitConditionPrivate::wait(unsigned long) + 43
4   org.qt-project.QtCore         	0x000000010f1a9df2 QWaitCondition::wait(QMutex*, unsigned long) + 162
[... 68 lines cut ...]

Thread 9 Crashed:: com.apple.audio.IOThread.client
0   ???                           	0x00000001179d0120 0 + 4691132704
1   com.apple.audio.units.Components	0x0000000117af1ea5 AUInputElement::PullInput(unsigned int&, AudioTimeStamp const&, unsigned int, unsigned int) + 185
2   com.apple.audio.units.Components	0x00000001179e8858 AUInputFormatConverter2::InputProc(OpaqueAudioConverter*, unsigned int*, AudioBufferList*, AudioStreamPacketDescription**, void*) + 196
3   com.apple.audio.toolbox.AudioToolbox	0x00007fff9c1ad5ab AudioConverterChain::CallInputProc(unsigned int) + 415
4   com.apple.audio.toolbox.AudioToolbox	0x00007fff9c1ad308 AudioConverterChain::FillBufferFromInputProc(unsigned int*, CABufferList*) + 130
5   com.apple.audio.toolbox.AudioToolbox	0x00007fff9c1ad26c BufferedAudioConverter::GetInputBytes(unsigned int, unsigned int&, CABufferList const*&) + 178
6   com.apple.audio.toolbox.AudioToolbox	0x00007fff9c1ad130 CBRConverter::RenderOutput(CABufferList*, unsigned int, unsigned int&, AudioStreamPacketDescription*) + 106
7   com.apple.audio.toolbox.AudioToolbox	0x00007fff9c1a076d BufferedAudioConverter::FillBuffer(unsigned int&, AudioBufferList&, AudioStreamPacketDescription*) + 281
8   com.apple.audio.toolbox.AudioToolbox	0x00007fff9c1a0a31 AudioConverterChain::RenderOutput(CABufferList*, unsigned int, unsigned int&, AudioStreamPacketDescription*) + 99
[... cut ...]
16  com.apple.audio.CoreAudio     	0x00007fff9cf0b8ce HALC_ProxyIOContext::IOWorkLoop() + 5290
17  com.apple.audio.CoreAudio     	0x00007fff9cf0a264 HALC_ProxyIOContext::IOThreadEntry(void*) + 128
18  com.apple.audio.CoreAudio     	0x00007fff9cf09fa3 HALB_IOThread::Entry(void*) + 75
19  libsystem_pthread.dylib       	0x00007fffb2d1e9af _pthread_body + 180
20  libsystem_pthread.dylib       	0x00007fffb2d1e8fb _pthread_start + 286
21  libsystem_pthread.dylib       	0x00007fffb2d1e101 thread_start + 13
```

AlbrechtL, 2017-03-31:
Please open an new issue for the crash problem.

AlbrechtL, 2017-03-31:
Compiling welle.io with QT 5.7 avoids thisproblem. Details can be found at https://bugreports.qt.io/browse/QTBUG-59040

closed 2017-03-31 by AlbrechtL