AlbrechtL/welle.io #36: [Suggestion] add a simple "recording" function

Wikinaut, 2017-03-26:
Perhaps this can be useful, now and then, to start and stop audio recording.

Wikinaut, 2017-03-26:
I "want" to have a record button in "welle.io", and transcoding to mp3 (in reasonable user-definable quality) would be fine, and I want to record *as* I listen.

AlbrechtL, 2017-03-27:
I don't see a real improvement to add this feature. You can do the same with a virtual sound card and some specific transcode applications.
It adds just a little bit more usability for this use case but creates a lot of work to develop and maintain it. I also think that regular users don't need this feature.

closed 2017-03-27 by AlbrechtL

lars18th, 2018-02-15:
Hi,

Only as a _reference_: If you need to **RECORD the IQ samples**, you can use another tool: "QIRX SDR" or the good known "qtdab". See more at: http://www.welle.io/devices/rawfile#odr-dabmod

Regards!