AlbrechtL/welle.io #37: [Suggestion] add a command line start parameter to start playing a certain station

Wikinaut, 2017-03-26:
[Suggestion] add a command line start parameter to start playing a certain station.

AlbrechtL, 2017-03-27:
Can you explain a little bit more why you need this feature?

Wikinaut, 2017-03-27:
I want to have a shortcut on my Linux desktop which when clicked starts playing "DeutschlandradioKultur".

AlbrechtL, 2017-03-28:
I agree that a feature is needed that starts the playing automatically. But I'm not sure if a command line option is a perfect approach.

Wikinaut, 2017-03-28:
Several shortcuts: a use of an ini file, for example, would not allow to have several shortcuts with different stations.

AlbrechtL, 2017-04-17:
I added your request to the [task list](https://github.com/AlbrechtL/welle.io/wiki/Open-Tasks).
If more people needs this feature they can write it here.

closed 2017-04-17 by AlbrechtL