AlbrechtL/welle.io #38: [Suggestion] GPS, Signal Strength,... Logfile + GPS base Switch + Transponder File

nousername1, 2017-03-27:
Hi
a) Pls add an GPS connection for (Win) Pcs to record in a logfile the current Signal strength, Bit Error,... everything what the Software could output.
It would be very nice to collect a lot of logfiles from user who use it onboard a Car or Train, Bus,... to create a Map where the Signal/ Transponder is.

b) When a Map is created it would be very cosy to use the GPS Position to switch between the different Transponder/ Channel before the Signal get weak. (For example in the Boarder Regions and or in the Mountain)

c) It also would be nice to load a Transponder File into the System. In the most case the Channel /Transponder list is very static it would be nice.

andimik, 2017-03-27:
For whole transponder see eti-stuff or dabtools (but this is only working with RTL Sticks) 

GPS logging is a good idea, but I have to find a laptop which does not destroy the signal when I switch it on.

nousername1, 2017-03-27:
Witch Signal? DAB+? I use an cheap Ebay Amateur Radio Antenna + cheap 1m Antenna Cable with an LNA.
The Antenna is outdoor and the LNA, Dongle indoor. That Setup work. I know the Antenna Cable is bend by the Window....
The Results are quite good.

AlbrechtL, 2017-03-27:
This is a quite interesting feature.

But I think this use case sounds more like a tasks for a specific script. 
You can use [dab-cmdline](https://github.com/JvanKatwijk/dab-cmdline) in combination with a GPS tool and than use a script language (bash, python, ruby, etc.) to control both programs.

I don't see this specific feature in welle.io because it is to special for the most of the regular users.

closed 2017-03-27 by AlbrechtL

nousername1, 2017-03-31:
dont work in Windows what I see.