AlbrechtL/welle.io #42: Bad graphic output (unstable, flickering)

AlbrechtL, 2017-03-30:
I received the following question by e-mail:

> The first problem is in terms of definition of the graphic fonts that appearing all unstable and flickering...
> 
> [image]
> 
> Second: the channel searchs never ends successfully and completely missing the sound of channels found.
> 
> You already know the problem, from other guys?
> I tried it on two different computers with my AirSpy R2 both with Windows7 32-bit.

welle.io uses OpenGL for the graphic output. I suppose there is some problem with your graphic card. Can you post the command line output?

Also the second problem can be related to the graphic problem because if you CPU load is to high due the graphic problem there is no space to decode DAB.

Which CPU and graphic card do you are using?

PiNoise, 2017-04-01:
HI AlbrechtL,
Thank for your attention and your commitment

* OpenGL-graphic output:  Ok, but until now I never had any problems with my graphic card. I use it long since (many radioamatorial software and sdr) without problems.
* command line output:  What is the name of the file?
* my CPU: Intel Core 2 Duo T8300 @ 2.40 GHz with 4 GBytes DDR2 RAM
* Graphics: embedded Intel 965 Express Chipset Family

On this Acer laptop running properly all my SDR software: as Perseus, WinradioG33DC, HDSDR, SDR# with AirspyR2 and equivalent DAB software as DabPlayer and SDR-J.

73, Paolo

andimik, 2017-04-02:
> command line output: What is the name of the file?

Open `cmd.exe`, type in `c:\Program Files (x86)\welle.io\welle-io.exe` (if you used the default installation path; hint, it's a minus in the exe-file, not a dot), then you have the command line output.

PiNoise, 2017-04-02:
many thanks, this is my initial "command line output" with AirSpy R2:

[Microsoft Windows [Versione 6.1.7601]
Copyright (c) 2009 Microsoft Corporation. Tutti i diritti riservati.
c:\radio\welledab>welle-io.exe
InputFactory: Input device: "auto"
Airspy: Open airspy
Airspy: 2 sample rates are supported
Airspy: sample rates: 0 10000000
Airspy: sample rates: 1 2500000
Airspy: selected samplerate 2500000
Audio: ActiveState
GUI: AGC on
qml: dpiScaleFactor: 1
GUI: AGC off
GUI: Starting true
OFDM-processor: start
OFDM-processor: closing down
OFDM-processor: start
Audio: StoppedState
OFDM-processor: closing down
OFDM-processor: start
GUI: "12B" -> 225648000
c:\radio\welledab>_

andimik, 2017-04-02:
Have you tried another theme or design in Windows?

Do you have a 2nd PC to test?

AlbrechtL, 2017-04-02:
Please try it also again with the [latest version](https://github.com/AlbrechtL/welle.io/releases/tag/V20170331) because we switched the QT version.

PiNoise, 2017-04-03:
HI !

In Window 7 I have one of the standard themes...

Yes, I have a 2nd PC with W7 Ultimate (32bit), Intel Core2 CPU 6600 @2.40 GHz, RAM 4,00 GB. 
But also here I have problems (I'm very unfortunate...), the file Exe don't starts with a generic Windows warning "WELLE-IO has stopped working..." (I enclose screen): 
(https://cloud.githubusercontent.com/assets/26836104/24620710/eedb7020-189e-11e7-8505-05cffcdff283.jpg)
Even with the latest version I do not get any improvement...   
I am very demoralized  :-((

73, Paolo

AlbrechtL, 2017-04-03:
This might be an OpenGL problem.

Please keep in mind that we don't support Windows 7 officially. We highly recommend to use a modern operating system like Windows 10 or Ubuntu 16.04 LTS.

PiNoise, 2017-04-04:
Yes, it will certainly be so. 
Thanks for timely support.
If I can, I'll try on a distro Linux Mint 17.3

AlbrechtL, 2017-04-04:
OK, I close this issue.
If you have trouble with Linux Mint 17.3 just open a new issue, please.

closed 2017-04-04 by AlbrechtL