AlbrechtL/welle.io #43: welle.io.pro won't build on Linux Mint 18.1 if airspy.h missing

stuzog, 2017-03-30:
Although airspy support is shown as an optional installation, welle.io.pro won't compile on Linux Mint 18.1 if airspy.h missing. Build terminates with:

```
g++ -c -pipe -g -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DHAVE_AIRSPY -DHAVE_RTLSDR -DQT_QML_DEBUG -DQT_QUICK_LIB -DQT_CHARTS_LIB -DQT_WIDGETS_LIB -DQT_MULTIMEDIA_LIB -DQT_GUI_LIB -DQT_QML_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -D_REENTRANT -I../welle.io -I. -I../welle.io -I../welle.io -I../welle.io/src -I../welle.io/src/ofdm -I../welle.io/src/backend -I../welle.io/src/backend/audio -I../welle.io/src/backend/data -I./src/backend/data/journaline -I../welle.io/src/output -I../welle.io/src/various -I../welle.io/src/input -I../welle.io/src/gui -isystem /usr/local/include -I../Qt/5.8/gcc_64/include -I../Qt/5.8/gcc_64/include/QtQuick -I../Qt/5.8/gcc_64/include/QtCharts -I../Qt/5.8/gcc_64/include/QtWidgets -I../Qt/5.8/gcc_64/include/QtMultimedia -I../Qt/5.8/gcc_64/include/QtGui -I../Qt/5.8/gcc_64/include/QtQml -I../Qt/5.8/gcc_64/include/QtNetwork -I../Qt/5.8/gcc_64/include/QtCore -I. -isystem /usr/include/libdrm -isystem /usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I../Qt/5.8/gcc_64/mkspecs/linux-g++ -o CInputFactory.o ../welle.io/src/input/CInputFactory.cpp
In file included from ../welle.io/src/input/CInputFactory.cpp:39:0:
../welle.io/src/input/CAirspy.h:44:30: fatal error: libairspy/airspy.h: No such file or directory
compilation terminated.
Makefile:12219: recipe for target 'CInputFactory.o' failed
make: *** [CInputFactory.o] Error 1
13:58:29: The process "/usr/bin/make" exited with code 2.
Error while building/deploying project welle.io (kit: Desktop Qt 5.8.0 GCC 64bit)
When executing step "Make"
13:58:29: Elapsed time: 00:01.
```

AlbrechtL, 2017-03-31:
Thanks for the hint. I updated the build how to [Ubuntu Linux 16.04 LTS](https://github.com/AlbrechtL/welle.io/blob/master/README.md#ubuntu-linux-1604-lts). Please read it again.

closed 2017-03-31 by AlbrechtL

stuzog, 2017-03-31:
Thanks, that worked.