AlbrechtL/welle.io #44: Build on OS X using Homebrew?

stuzog, 2017-03-30:
Is it possible to build well.io.pro without having to install MacPorts? I prefer Homebrew but Homebrew can't find fftw-3-single. Proceeding without it results in the expected compile error:

`../src/various/fft.h:42:10: fatal error: 'fftw3.h' file not found`

AlbrechtL, 2017-03-31:
Unfortunately, we only support MacPorts. Please follow [this](https://github.com/AlbrechtL/welle.io#macos) how to to compile welle.io for macOS.

closed 2017-03-31 by AlbrechtL

kgarrels, 2017-03-31:
have you tried http://brewformulas.org/Fftw ?

Otherwise you probably need to build from source.

stuzog, 2017-04-01:
@kgarrels @AlbrechtL Thanks for the hint. `brew` will install `fftw`, `faad2`, `librtlsdr` and `libusb`, but not `fftw-3-single` , which is the only one I can't find to install.

HansVanEijsden, 2017-04-13:
@stuzog I have it working (MacOS Sierra) with homebrew. Very simple!

First of all: `brew install fftw faad2 rtl-sdr libusb airspy`
This will install all you need in /usr/local/include and /usr/local/lib.

Then you need to change two paths in welle.io.pro (line 158), changing opt for usr:
`macx {
    INCLUDEPATH	+= /usr/local/include
    LIBS    += -L/usr/local/lib`

[image]
Enjoy! 😃

stuzog, 2017-04-13:
Thanks @HansVanEijsden I made the change but Build still stalls on not finding fftw3 (screen shot attached). `fftw-3.3.6-pl2` was installed by `brew`.

[image]

HansVanEijsden, 2017-04-13:
That's strange @stuzog - can you please check if fftw3.h is in `/usr/local/include`?
[image]
And make sure your welle.io.pro looks like mine:
[image]

stuzog, 2017-04-13:
Hi @HansVanEijsden -- yes, it is:

```
bash$ lsla /usr/local/include/fftw3.*
lrwxr-xr-x 1 Stuart admin 40 1 Apr 12:05 /usr/local/include/fftw3.f -> 
../Cellar/fftw/3.3.6-pl2/include/fftw3.f
lrwxr-xr-x 1 Stuart admin 42 1 Apr 12:05 /usr/local/include/fftw3.f03 -> 
../Cellar/fftw/3.3.6-pl2/include/fftw3.f03
lrwxr-xr-x 1 Stuart admin 40 1 Apr 12:05 /usr/local/include/fftw3.h -> 
../Cellar/fftw/3.3.6-pl2/include/fftw3.h
bash$
bash$ lsla /usr/local/Cellar/fftw/3.3.6-pl2/include/
total 296
drwxr-xr-x 7 Stuart admin 238 27 Jan 13:14 .
drwxr-xr-x 15 Stuart admin 510 13 Apr 09:13 ..
-rw-r--r-- 1 Stuart admin 2447 27 Jan 13:14 fftw3.f
-rw-r--r-- 1 Stuart admin 54596 27 Jan 13:14 fftw3.f03
-rw-r--r-- 1 Stuart admin 31394 27 Jan 13:14 fftw3.h
-rw-r--r-- 1 Stuart admin 26983 27 Jan 13:14 fftw3l.f03
-rw-r--r-- 1 Stuart admin 25682 27 Jan 13:14 fftw3q.f03
bash$
```

And `welle.io.pro` looks exactly the same.

Note that I'm running El Capitan 10.11.6. on a 2007 iMac 7,1.

HansVanEijsden, 2017-04-13:
That's really strange, @stuzog - here I've got exactly the same files with the same sizes. I can build it without problems on my iMac and MacBook Pro.
[welle-io.app.zip](https://github.com/AlbrechtL/welle.io/files/920372/welle-io.app.zip)
Here is my compiled app. You can try it if you like, but of course it's dynamically linked and it needs all the dependencies. Let me know if it works.

stuzog, 2017-04-13:
@HansvanEijsden This is where it breaks, I think:

HansVanEijsden, 2017-04-13:
@stuzog your phasereference.cpp looks different than mine. Are you sure you have a fresh source tree? Try doing `git pull` of do a fresh `git clone`?
This is mine:
[image]

stuzog, 2017-04-13:
@HansvanEijsden I had done a fresh `git clone` earlier today but a `git pull` changed 64 files. However, same issue, here's the Compile output of a rebuild:

```
10:50:19: Running steps for project welle.io...
10:50:19: Starting: "/usr/bin/make" clean
rm -f qrc_touch_gui_resource.cpp
rm -f moc_predefs.h
rm -f moc_ofdm-processor.cpp moc_ofdm-decoder.cpp moc_fic-handler.cpp moc_fib-processor.cpp moc_mp2processor.cpp moc_mp4processor.cpp moc_faad-decoder.cpp moc_pad_decoder_adapter.cpp moc_CAudio.cpp moc_CRTL_TCP_Client.cpp moc_CStationList.cpp moc_CGUI.cpp moc_CRadioController.cpp moc_CAirspy.cpp moc_CRTL_SDR.cpp
rm -f main.o ofdm-processor.o ofdm-decoder.o phasereference.o phasetable.o freq-interleaver.o viterbi.o fic-handler.o msc-handler.o protection.o eep-protection.o uep-protection.o fib-processor.o galois.o reed-solomon.o charsets.o firecode-checker.o dab-virtual.o dab-processor.o protTables.o dab-audio.o mp2processor.o mp4processor.o pad_decoder.o mot_manager.o pad_decoder_adapter.o tools.o CAudio.o fft.o Xtan2.o CInputFactory.o CNullDevice.o CRAWFile.o CRTL_TCP_Client.o DabConstants.o CMOTImageProvider.o CStationList.o CGUI.o CRadioController.o CChannels.o CAirspy.o CRTL_SDR.o qrc_touch_gui_resource.o moc_ofdm-processor.o moc_ofdm-decoder.o moc_fic-handler.o moc_fib-processor.o moc_mp2processor.o moc_mp4processor.o moc_faad-decoder.o moc_pad_decoder_adapter.o moc_CAudio.o moc_CRTL_TCP_Client.o moc_CStationList.o moc_CGUI.o moc_CRadioController.o moc_CAirspy.o moc_CRTL_SDR.o
rm -f *~ core *.core
10:50:19: The process "/usr/bin/make" exited normally.
10:50:19: Configuration unchanged, skipping qmake step.
10:50:19: Starting: "/usr/bin/make" 
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -c -pipe -stdlib=libc++ -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk -mmacosx-version-min=10.9  -g -std=gnu++11 -Wall -W -fPIC -DHAVE_AIRSPY -DHAVE_RTLSDR -DQT_QML_DEBUG -DQT_QUICK_LIB -DQT_CHARTS_LIB -DQT_WIDGETS_LIB -DQT_MULTIMEDIA_LIB -DQT_GUI_LIB -DQT_QML_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -Isrc -Isrc/ofdm -Isrc/backend -Isrc/backend/audio -Isrc/backend/data -Isrc/backend/data/journaline -Isrc/output -Isrc/various -Isrc/input -Isrc/gui -I/opt/local/include -I../../Qt/5.8/clang_64/lib/QtQuick.framework/Headers -I../../Qt/5.8/clang_64/lib/QtCharts.framework/Headers -I../../Qt/5.8/clang_64/lib/QtWidgets.framework/Headers -I../../Qt/5.8/clang_64/lib/QtMultimedia.framework/Headers -I../../Qt/5.8/clang_64/lib/QtGui.framework/Headers -I../../Qt/5.8/clang_64/lib/QtQml.framework/Headers -I../../Qt/5.8/clang_64/lib/QtNetwork.framework/Headers -I../../Qt/5.8/clang_64/lib/QtCore.framework/Headers -I. -I../../Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/System/Library/Frameworks/OpenGL.framework/Headers -I../../Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/System/Library/Frameworks/AGL.framework/Headers -I../../Qt/5.8/clang_64/mkspecs/macx-clang -F/Applications/Qt/5.8/clang_64/lib -o main.o src/main.cpp
In file included from src/main.cpp:44:
In file included from src/CRadioController.h:41:
In file included from src/ofdm/ofdm-processor.h:40:
In file included from src/ofdm/phasereference.h:28:
src/various/fft.h:42:10: fatal error: 'fftw3.h' file not found
#include        <fftw3.h>
                ^
1 error generated.
make: *** [main.o] Error 1
10:50:23: The process "/usr/bin/make" exited with code 2.
Error while building/deploying project welle.io (kit: Desktop Qt 5.8.0 clang 64bit)
When executing step "Make"
10:50:23: Elapsed time: 00:04.
```

I'm wondering if it's to do with the Version 8.2.1 (8C1002) of Xcode that is running?

HansVanEijsden, 2017-04-13:
> I'm wondering if it's to do with the Version 8.2.1 (8C1002) of Xcode that is running?
That's a good possibility. Here I have Sierra 10.12.4 (16E195) on all devices. My Xcode version is 8.3.1 (8E1000a). What about my .app (the link 4 replies above)? Does that work?

kgarrels, 2017-04-13:
I don’t think so…I used 8.3…beta.

can you show us:

``
ls -l /usr/local/include/ff*
``
?

It really looks like the include file is not there.

kai

stuzog, 2017-04-13:
@HansvanEijsden The version you sent crashed immediately. I just updated 
my last comment to show I'm running `clang` 8.0.0 and XCode Version 
8.2.1 (8C1002).

stuzog, 2017-04-13:
@kgarrels

```
bash$ ls -l /usr/local/include/ff*
lrwxr-xr-x 1 Stuart admin 40 1 Apr 12:05 /usr/local/include/fftw3.f -> 
../Cellar/fftw/3.3.6-pl2/include/fftw3.f
lrwxr-xr-x 1 Stuart admin 42 1 Apr 12:05 /usr/local/include/fftw3.f03 -> 
../Cellar/fftw/3.3.6-pl2/include/fftw3.f03
lrwxr-xr-x 1 Stuart admin 40 1 Apr 12:05 /usr/local/include/fftw3.h -> 
../Cellar/fftw/3.3.6-pl2/include/fftw3.h
lrwxr-xr-x 1 Stuart admin 43 1 Apr 12:05 /usr/local/include/fftw3l.f03 
-> ../Cellar/fftw/3.3.6-pl2/include/fftw3l.f03
lrwxr-xr-x 1 Stuart admin 43 1 Apr 12:05 /usr/local/include/fftw3q.f03 
-> ../Cellar/fftw/3.3.6-pl2/include/fftw3q.f03
bash$
```

HansVanEijsden, 2017-04-13:
@stuzog thanks for trying, too bad it doesn't run there (not surprisingly). 

My clang:
```
hans$ clang -v
Apple LLVM version 8.1.0 (clang-802.0.41)
Target: x86_64-apple-darwin16.5.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```
My Qt, the same as yours:
[image]

@kgarrels thanks for your great work by porting it to MacOS. Really appreciated.

stuzog, 2017-04-13:
@HansvanEijsden @kgarrels Thanks for the assistance. This iMac is a late-2007 early aluminium 24" that may have issues that later models may have fixed. It has failed to install one or two other programs, although that could be me. Sigh.... obsolescence creeping in to both my iMac and I?

kgarrels, 2017-04-13:
ok, I am not used to home-brew...seems the files are in the cellar :)

Should it be that way? Are they really there?

Maybe you copy them directly to /usr/local/include and try again.

I am now away for two weeks, then we can look again.

Ciao,
kai

kgarrels, 2017-04-13:
I have the same clang version.

kai