AlbrechtL/welle.io #45: welle.io cannot access the hardware when running the AppImage

probonopd, 2017-03-31:
Let's make it super simple for AppImage users to unload the kernel module and install a udev rule.

Please see https://github.com/JvanKatwijk/qt-dab/issues/34, especially https://github.com/JvanKatwijk/qt-dab/issues/34#issuecomment-289270761 for an idea - do you have a better one?

AlbrechtL, 2017-03-31:
Are qt-dab and welle.io the first programs that are using hardware from an AppImage?

Another solution is to use the system librtlsdr and the AppImage is using it. The disadvantage is that the user has to install it before using the AppImage.

Are they any ways to install an AppImage permanently? If yes, the AppImage can install the udev rule.

probonopd, 2017-03-31:
> Are qt-dab and welle.io the first programs that are using hardware from an AppImage?

Well, if with "using hardware" you mean "need root access to /dev/*`, then yes.

> Another solution is to use the system librtlsdr and the AppImage is using it. The disadvantage is that the user has to install it before using the AppImage.

Yes, that would be very inconvenient for the user, we should avoid it.

> Are they any ways to install an AppImage permanently? If yes, the AppImage can install the udev rule.

You don't "install" an AppImage, you just execute it. The payload (=what is inside the application) should then take care to do everything that is needed (i.e., become root and install udev rules, etc.)

Hence the idea with my scripts.

AlbrechtL, 2017-04-17:
I don't know which is the best solution.
Please send me a pull request with a solution that you are fine with.

AlbrechtL, 2017-05-23:
I'm closing this issue for now because no one complained about this problem

closed 2017-05-23 by AlbrechtL