AlbrechtL/welle.io #47: strange problem

zygmund2000, 2017-04-11:
Hi,

I have to machines with identical system Majnaro 64 bit, all packages are identical but on one of them I get this error: Project ERROR: Unknown module(s) in QT: charts multimedia
I don't know how to fix that, looks like something missing but I checked everything.
Maybe somebody know how...

Regards

AlbrechtL, 2017-04-11:
QT charts and QT multimedia are missing on your system. Please install it.

zygmund2000, 2017-04-11:
I wrote earlier that I have everything installed, so all means all - also QT charts and QT multimedia.

AlbrechtL, 2017-04-12:
I suppose your QT installation is broken. Please use the QT installer from https://www.qt.io/download-open-source and install it again.

closed 2017-04-12 by AlbrechtL