AlbrechtL/welle.io #48: Display AAC profile used in expert mode

hairmare, 2017-04-12:
Thanks for creating the first sdr based receiver that I was able to compile on my mac! 

I was wondering what AAC profile the stations I'm receiving with my rtl_sdr dongle are using and ended up doing a quick proof of concept hack by adding the following line at the end of the [`fib_processor::bind_audioService`](https://github.com/AlbrechtL/welle.io/blob/0eb4a3f87bc6a3b781359eafedfe5284c9d9f8e6/src/backend/fib-processor.cpp#L980-L1008) function in `fib-processor.cpp`.

```
        qDebug() << "PS Flag=" << components[firstFree].PS_flag;
```

As expected all the stations I was able to receive in Bern, Switzerland had the PS flag set. Since PS  seems to be unique to the aacPlus v2 profile I'm reasonably sure that everyone is using it given the debug output from my change.

I think displaying the used profile would be a nice enhancement to the expert mode but my C++/Qt fu isn't strong enough to write a proper patch. Would you consider adding such a feature? I think dablin displays this info, maybe it has more logic to really figure out the exact profile.

AlbrechtL, 2017-04-12:
Can you explain in detail why you need to know the used AAC profile?

hairmare, 2017-04-12:
I was discussing this with our broadcast techie and it turned out that we were not sure if v2 had already been rolled out everywhere. I figured this info would be handy for anyone setting up their own dab encoder as a means to check if the expected profile is being using in an end to end test fashion.

andimik, 2017-04-12:
For such reasons please use the freeware of Andreas Gsinn from ukwtv.de forum (which you already know probably). The Realtek driver has it implemented.

hairmare, 2017-04-12:
I didn't know that one yet. I'm assuming you are talking about [this](http://www.ukwtv.de/cms/downloads-aside/281-dab-player-von-andreas-gsinn.html)?

It looks like it's windows only and we mainly use Linux and some macOS. I might be able to get it working in WINE though.

hairmare, 2017-04-12:
I gave Andreas' player a try and was able to get it working after installing the proper driver for my R820T tuner and manually starting rtl_tcp. The channel scan was only able to detect a single ensemble where welle.io sees 4 ensembles. I assume this is due to a longer dwell time on individual channels as well as doing frequency correction during the scan. On the ensemble I'm seeing it does display the used profile.

Preferably I'd like to use open/libre software that also runs natively on Linux and macOS.

FWIW, the code that does the profile detection in dablin seems to be [here](https://github.com/Opendigitalradio/dablin/blob/ca6a3688b1f46586c4764e0e65627794576b2d1c/src/dabplus_decoder.cpp#L233).

andimik, 2017-04-12:
And why not take a look into FMLIST resp. DABLIST? The ensembles from Switzerland are well documented. 

http://www.fmlist.org/sendertabelle/dab-ww.php

hairmare, 2017-04-12:
I was just looking at that, turns out 12C (SRG SSR F01) which I can receive with both DAB Player and welle.io should be transmitting from the same site as 8B (SMC D03 BE-FR) which I can receive only with welle.io. I'm assuming that I should be getting a signal from [Bantiger](http://fmscan.org/transmitter.php?i=7300039) but I'll have to try it out at a location with a proper line of sight to the transmitter to be sure.

I also tried DAB Player in antenna alignment mode and was unable to get an SNR above 6 db (even when tuning to 12C which both can receive). With welle.io I'm getting an consistent SNR of approx. 18 dB.

I'm using the latest git commit (e3e6ee2) of rtl_sdr.

andimik, 2017-04-12:
This is a common problem with Linux. 

Try it under Windows and you will have much better reception!

hairmare, 2017-04-12:
I feel like this is getting off-topic. Maybe I should move this discussion to crc-mmbtools or somewhere else.

I gave welle.io a try with `-D rtl_tcp` and while it picks up on most of the ensembles (with the exception of 12A which almost made me feel like a DXer anyhow). I can confirm that I'm not getting usable SNR values with welle.io in `rtl_tcp` mode (only slightly higher than what I saw with DAB Player, but not enough to tune to 12C having only a 8 dB SNR).

I haven't used windows in ages and will need to check if I can find a PC to test that. BTW, I'm currently on macOS and not Linux.

andimik, 2017-04-12:
Not really off-topic, I just wanted to help you to get your answers ;-) 

Liebe Grüße in die Schweiz

hairmare, 2017-04-12:
I'm spinning up some IE VMs to get a virtual Windows to test against.

Vielen Dank für die Hilfe und liebe Grüsse nach Österreich :)

hairmare, 2017-04-12:
I managed to get a Win10 vm up and running but I'm seeing lots of Frame and FIC CRC errors and not getting any useable audio out of welle.io (though it does report the same SNR as on a mac). DAB Player completely fails to detect my RTL SDR dongle.😞

I'll have to try this on a non virtualized box as I'm assuming that the USB passthrough in VirtualBox is the issue with this setup.

AlbrechtL, 2017-04-12:
Guys, I would like to bring you back to the origin topic.

I added your wish to see the AAC profile to the [task list](https://github.com/AlbrechtL/welle.io/wiki/Open-Tasks). But please keep in mind that it is actually not planned to add this into welle.io. If more people needs this feature they can write it into this issue.

closed 2017-04-12 by AlbrechtL

kgarrels, 2017-04-12:
yes, VirtualBox USB performance on macOS is not nice...VMware Fusion is a lot better.
guest Windows version seems to play a role too, I am on Windows 7 still for this kind of stuff.
 

> I'll have to try this on a non virtualized box as I'm assuming that the USB passthrough in VirtualBox is the issue with this setup.

HansVanEijsden, 2017-04-16:
> If more people needs this feature they can write it into this issue.

Yes, please. Currently there are 3 stations here audible with very high pitched audio and with dropouts (11B, The Netherlands, MTVNL). Another station on that same ensemble sounds fine. I suspect it's the difference in AAC profile but I'm not sure. That's why I would love to see the used AAC profile too.

AlbrechtL, 2017-04-16:
How does Qt-DAB performs? Same issue there?

HansVanEijsden, 2017-04-16:
@AlbrechtL unfortunately I can't test it, because the only Linux machines I have are my servers in the datacenters. At home all my machines are on MacOS 10.12.4 (Sierra).

Here some information about the versions of the dependencies on my system, maybe it helps:
- fftw: stable 3.3.6-pl2
- faad2: stable 2.7
- librtlsdr: stable 0.5.3
- libusb: stable 1.0.21
- airspy: stable 1.0.9 (don't know if that's getting used by welle.io - I didn't enable it)

Let me know if you need audio recordings and/r if you want me to open a new issue.

AlbrechtL, 2017-04-16:
Yes, please open a new issue and give as much as possible details (e.g. command line output, OS version, screenshot incl. expert mode). It can be also a sample rate issue.

HansVanEijsden, 2017-04-16:
Thank you! Done that, see https://github.com/AlbrechtL/welle.io/issues/51