AlbrechtL/welle.io #51: Some stations: high pitched audio with dropouts

HansVanEijsden, 2017-04-16:
Currently there are some stations here audible with very high pitched audio and with dropouts. Another station on that same ensemble sounds fine. I suspect it's the difference in AAC profile but I'm not sure. 
It's ensemble 11B (MTVNL), transmitter Markelo, The Netherlands. The reception here in Zwolle is weak but good.

Here is some information about the versions of the dependencies on my system, maybe it helps:
- fftw: stable 3.3.6-pl2
- faad2: stable 2.7
- librtlsdr: stable 0.5.3
- libusb: stable 1.0.21
- airspy: stable 1.0.9 (don't know if that's getting used by welle.io - I didn't enable it)

System info:
```$ uname -a
Darwin iMac-van-Hans-van-Eijsden.local 16.5.0 Darwin Kernel Version 16.5.0: Fri Mar  3 16:52:33 PST 2017; root:xnu-3789.51.2~3/RELEASE_X86_64 x86_64
```
Newest MacOS Sierra, 10.12.4 (16E195). Newest welle.io from git, compiled with this Qt Creator version:
[image]
Here's NH Radio, sounding good without problems:
[NH_Radio_Good.mp3.zip](https://github.com/AlbrechtL/welle.io/files/924501/NH_Radio_Good.mp3.zip)
[image]
[image]
Here's HotRadio Hits, sounding bad with high pitch and dropouts:
[HotRadio_Hits_Bad.mp3.zip](https://github.com/AlbrechtL/welle.io/files/924502/HotRadio_Hits_Bad.mp3.zip)
[image]
[image]
And here's BBC Worldservice, also sounding bad with that high pitch and those dropouts:
[BBC_Worldservice_Bad.mp3.zip](https://github.com/AlbrechtL/welle.io/files/924503/BBC_Worldservice_Bad.mp3.zip)
[image]
[image]
Information about the problematic ensemble: http://radio-tv-nederland.nl/dab/mtvnl-11b.html
All the other > 50 stations here are sounding perfect. Just let me know if you need more info. I'm happy to help!

HansVanEijsden, 2017-04-16:
Update, thanks to user [Badeendje @ the Dutch Radio Forum](https://www.radioforum.nl/viewtopic.php?p=452558#p452558): the problematic stations are transmitting with a sample rate of **32kHz**. All the other stations are transmitting with a sample rate of **48kHz**.

So, there we have it I think: 32kHz stations are giving a high pitch and dropouts.

HansVanEijsden, 2017-04-16:
I've made a 20 seconds raw recording of the 11B multiplex.

```
$ ./dab_raw_record.sh -c 11B -t 20

Run rtl_sdr -f 218640000 -s 2048000 -g 42 -n 40960000 20170416_224242_11B.iq
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000004

Using device 0: Generic RTL2832U OEM
Found Elonics E4000 tuner
Sampling at 2048000 S/s.
Tuned to 218640000 Hz.
Tuner gain set to 42.00 dB.
Reading samples in async mode...
```

Download the raw .iq file here: https://www.hansvaneijsden.nl/temp/20170416_224242_11B.iq

I can open the file with this command:
`$ ./welle.io/welle-io.app/Contents/MacOS/welle-io -D rawfile -F 20170416_224242_11B.iq -B u8`

Then you can tune to the different stations. You can hear NH Radio with good audio (48kHz station) and BBC Worldservice, LX Classics, Efteling, West, HotRadio and GrootNieuws with bad audio (32kHz stations).
I hope this file makes testing easier for you. 😃

AlbrechtL, 2017-04-17:
Thanks for the detailed bug description and the test RAW file.
I fixed the issue in 0094073. Please test it again and report if the sound is OK now.

BTW: Your RAW file is showing another problem. The SNR calculation is not working properly because there are strong neighbour channels available. I opened a new issue #52 for this problem.

HansVanEijsden, 2017-04-17:
Wow! The issue is completely fixed now. Thank you very much for the (really quick) fix!
And thanks for noticing the other issue about the SNR, I noticed it too but I thought the signal was too weak to do a proper calculation. Happy to see I was wrong. 😉

closed 2017-04-17 by AlbrechtL