AlbrechtL/welle.io #52: SNR calculation not working if strong neighbour channels available

AlbrechtL, 2017-04-17:
The SNR calculation is not working if strong neighbour channels are available.
[image]

mpbraendli, 2018-04-05:
SNR could be calculated by comparing energy of the PRS and data symbols against energy of the NULL symbol, instead of looking at out-of-band power.

AlbrechtL, 2018-04-05:
Good idea.
But from my understanding we have to be synchronized to calculate the SNR. So we can use the current SNR calculation while there is no sync and use the new one if we are in sync.

AlbrechtL, 2020-09-30:
Fixed in https://github.com/AlbrechtL/welle.io/commit/96208fce17dabcadc200df2c97d455717d97db98.
[image]

closed 2020-09-30 by AlbrechtL