AlbrechtL/welle.io #54: How To install on MACOsx

djnutmac, 2017-04-18:
Dear DAB listeners, I would like to install Welle.io on my MACOSx, but I don't know how to start from, I'm not addicted to linux line commands, only copy and paste, is there any step-by-step guide?

Thankyou!

HansVanEijsden, 2017-04-18:
Hi @djnutmac, I will try to help you in simple steps.

- Open Terminal. Commands are followed by a return/enter.
- [Terminal] First we install Homebrew. Paste this command: `/usr/bin/ruby -e "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install)"`
- [Terminal] Now we are going to install the dependencies with this command: `brew install fftw faad2 rtl-sdr libusb airspy git`
- If you get a notice about installing Xcode: install Xcode, following the instructions. Open Xcode and install the command line tools. Close Xcode.
- Install Qt Creator. Download it at https://www.qt.io/download/ and choose Qt 5.7 in the installer options.
- [Terminal] download welle.io: `git clone https://github.com/AlbrechtL/welle.io.git`
- Open Qt Creator and open the project file `welle.io.pro`. Make sure in Qt Creator, "Projects, Build&Run, Run" that the checkbox "Add build library path to DYLD..." is off.
- Then you need to change two paths in welle.io.pro (line 158), changing "opt" for "usr":
`macx { INCLUDEPATH += /usr/local/include LIBS += -L/usr/local/lib`
- Build the program with the lower left icon in the Qt Creator. You can also run it directly with the green triangle.
- Enjoy!

AlbrechtL, 2017-04-18:
@HansVanEijsden, Are you interested in creating an installer for macOS?
For the Windows installer I used the QT Install Framework (QTIFW). Maybe these files are helping (https://github.com/AlbrechtL/welle.io-win-libs/tree/master/install).

HansVanEijsden, 2017-04-18:
@AlbrechtL I would love to do that! First I have to read more about it, so I know how it works. Currently it's a busy week here, but as soon as I can make time for it and as soon as I have some good results I'll let it know here.

AlbrechtL, 2017-04-18:
Great to hear! I'm looking forward to it.

djnutmac, 2017-04-18:
after first copY/paste, on terminal, nothing happens, no advise, nothing, what I have to do?

HansVanEijsden, 2017-04-18:
@djnutmac forgive my if my question is stupid, but..: did you press <enter> after pasting the command?

djnutmac, 2017-04-18:
yes!, I made the second command, and it says updating homebrew, maybe yet installed?

HansVanEijsden, 2017-04-18:
That's a good possibility. If it doesn't work, then wait for the installer (if I manage to read about it and to make one). I will do it as soon as I can make time for it. 😃

djnutmac, 2017-04-18:
I opened, I try to search this
Open Qt Creator and open the project file welle.io.pro. Make sure in Qt Creator, "Projects, Build&Run, Run" that the checkbox "Add build library path to DYLD..." is off.

But says
: error: Xcode not set up properly. You may need to confirm the license agreement by running /usr/bin/xcodebuild.

HansVanEijsden, 2017-04-18:
In point 4 I said:

> If you get a notice about installing Xcode: install Xcode, following the instructions. Open Xcode and install the command line tools. Close Xcode.

You can find Xcode in your Applications folder, or open it by typing "Xcode" in Spotlight and press Enter. Then you will receive a popup with the license agreement and it will install the remaining command line tools.

djnutmac, 2017-04-18:
mmm I think it's not installed, so I have to find old version for El Capitain because Xcode 8.0 does not work here....

HansVanEijsden, 2017-04-18:
Aaaaaahh El Capitan.. I don't know if I can help you with that, because there's a possibility some tools are different and/or Homebrew has different versions which do or do not work with welle.io. Here I'm testing with the newest MacOS Sierra 10.12.4 (16E195).

djnutmac, 2017-04-18:
ok, there's a different possibility to install on Linux? I have VMware Linux mint I use with tetra decoding, is it possible to follow same commands?

rensd, 2017-04-24:
Hi,

i got welle-io running on a 2008 MacPro with MacOS 10.11.6. Using macport and Qt 5.7.
Xcode 8.2.1. Just followed the information on the github README.md. The only problem
I have is the slideshow not sure if this broadcasted by any of the Dutch stations.

The SDR I use is an RTL-SDR dongle.

Regards Rens

rensd, 2017-04-24:
[image]

ubergeek23, 2017-04-28:
Ok, i got everything installed and compiled and it works!

Sound is a little choppy at times but i think that has to do with the application eating 80-90% of my Macbook's (Mid 2014 Retina) CPU-power. I know this is still in development but is there anything i can do when compiling this to make it eat less CPU?

AlbrechtL, 2017-04-30:
I'm closing this issue because all questions are answered and I'm looking forward for a macOS installer the makes it easy to install welle.io on Mac.

closed 2017-04-30 by AlbrechtL

blogmywiki, 2017-07-27:
I tried to follow the @HansVanEijsden instructions very closely to get welle.io working on OS X but I get the following error in the console when I run:
`Airspy: airpsy_open () failed: AIRSPY_ERROR_NOT_FOUND ( -5 )`
Airspy **is** installed. I changed the paths from 'opt' to 'usr'. Any ideas? Many thanks.

kgarrels, 2017-09-17:
Did you install air spy with macports?
Try "sudo port install airspy"
Change the paths back to opt before trying.