AlbrechtL/welle.io #55: sdr files from qt-dab not working

andimik, 2017-04-18:
Dear Albrecht,

I made some sdr-recordings in Südtirol this weekend. But they are not working in welle.io 

> ./welle-io -D rawfile -F /home/andreas/10c_dabmedia.sdr -B s16le

[image]
and I even get a seg fault here

Although sdr files seem to be s16le, I tried to load it as u8

> ./welle-io -D rawfile -F /home/andreas/10b_ras.sdr -B u8

Then you see a kind of spectrum (10B and 10D are used in Südtirol as well)
[image]

I had to convert them to rawfiles:

> sox -e unsigned-integer -r 240000 -t raw -c 1 -b 16 /home/andreas/10c_dabmedia.sdr -t raw -e signed-integer -b 8 -r 240000 /home/andreas/10c_dabmedia.raw

[image]

Only then it worked.

---

Can you check if welle.io expects 

unsigned-integer 2,4 MHz raw with 1ch and 16bit 

as `-B s16le`?

AlbrechtL, 2017-04-18:
Can you provide the RAW file?

andimik, 2017-04-18:
no, not again. 

Sending you such files exeeds my monthly free contingent.

Maybe you try yourself recording such a file with qt-dab and try to play them in welle.io.

HansVanEijsden, 2017-04-18:
@andimik you can use [WeTransfer.com](https://wetransfer.com) as free storage for large files up to 2GB per file. Just choose "send as link" at the options at the bottom.
[image]

andimik, 2017-04-18:
I know, but you didn't understand me, I'm talking about my provider's contingent, not storage. 

Furthermore, I really want to avoid sending sdr or raw files as my description clearly states how to create such files. In the past I only used them when there were problems with the content (for instance Protection Level 4A used in Slovenia, which did not work in qt-dab at all), not with a filetype definition issue.

AlbrechtL, 2017-04-18:
I understand your difficulties.
To reproduce a problem the easiest would be to use the same environment. In this case it is your file because you ran into the problem with this file. To save bandwidth you can also split the file and send only 100 MB.

andimik, 2017-04-18:
10 MB should be enough, this file is working in qt-dab and eti-stuff

https://we.tl/a4cEesw0i1

AlbrechtL, 2017-04-18:
The bytes inside your file are mixed. It seems the endianness is different. Did you record the file with the same set up like the last one?

andimik, 2017-04-18:
Good question ... I've updated qt-dab in the meanwhile.

But I tried two other qt-dab versions which I still have on my HDD from February and March, same result.

AlbrechtL, 2017-04-19:
Interesting, I have a SDR RAW file from another person with a different byte order. Anyway, I added a new RAW file format "s16be" (signed int 16 bit big endian).
Please test it again and report.

AlbrechtL, 2017-04-19:
Fixed in fd71e9f.

closed 2017-04-19 by AlbrechtL

andimik, 2017-04-19:
**Seems DABSticks produce s16le sdr files and Airspy this (new implemented) s16be.** 

I remember that a mux in Südtirol did not work with Airspy (Antenna problem??) and I used the DABStick instead. And this sdr works!

I confirm, that the s16be files recorded with Airspy under qt-dab work.

[image]