AlbrechtL/welle.io #57: Volume slider

ehthe, 2017-04-29:
Hi, first I wanted to thank you. welle.io is awesome !
My suggestion is to add a volume slider and optionally a mute button.

Use case : Change the volume independently of the system's mixer. For example pulseaudio does not remember the app. It gets a new name every launch like QtPulseAudio:4567. Thus you can't currently adjust the sound across runs.

Similar problem : https://sourceforge.net/p/wsjt/mailman/message/34166846/
QT issue : https://bugreports.qt.io/browse/QTBUG-32842

AlbrechtL, 2017-04-30:
I added the feature to the [task list](https://github.com/AlbrechtL/welle.io/wiki/Open-Tasks). But please keep in mind that it is currently not planned to add this into welle.io because it is only relevant for pluseaudio users.

I think it has to be fixed in QT. Please leave a message inside the [QT issue](https://bugreports.qt.io/browse/QTBUG-32842) that the QT developers increases the priority of this issue.

If more people needs this feature they can write it into here.

closed 2017-04-30 by AlbrechtL

mqus, 2017-06-20:
I have another problem with a similar solution: in any other application I can pause/mute the current stream (if you have to call someone, want to watch a short video on youtube etc.) Currently I can only quit the application entirely to make it stop emitting sounds, so it would really help if there was a volume slider/mute-button to temporarily disable the sound. Timeshifting(pause+continue) would also solve it but that would add another level of complexity (buffering the stream), which is propably not needed ;)

Aside from that, thank you for this simple-to-setup, easy-to-use dab-tuner! I'm using it on archlinux with rtl-sdr and it runs flawlessly.

gvanem, 2017-06-29:
I just discovered and installed Welle-io. I also have sound problem; i.e. not sound *at all*. I assume this is a Qt5 problem. The trace gives no indication of a sound problem:
```
Audio: ActiveState
RadioController: Number of gain steps: 28
RadioController: AGC on
qml: dpiScaleFactor: 1
qml: desktopAvailableWidth: 1920
qml: desktopAvailableHeight: 1050
qml: orientation: 2
file:///F:/gv/dx-radio/Welle-IO/qml/QtQuick/Controls/SplitView.qml:624: TypeError: Cannot call method 'updateFillIndex' of null
RadioController: Start or restart device
RadioController: Tune to channel "12B" -> 225.648 MHz
OFDM-processor: start
RadioController: Tune to station "NRK P1+Hordaland"
```
(but then nothing).

Looking at the sources, I assume the `CAudio` class is simply a thin wrapper for Qt's Multimedia library.

I use Win-10, RTL_SDR (Rafael Micro R820T tuner). 
*PS*. SDR-Sharp (v 1577) works fine.

AlbrechtL, 2017-06-29:
@gvanem Please very that all three rectangle are green in the right above corner.
[image]
More information: https://www.welle.io/faq

If the one of them is not green than you don't have sound.

gvanem, 2017-06-29:
Thanks. I briefly got sound for a few seconds. Seems my dongle/antenna is not up for the range > 200 MHz 🤕 .

andimik, 2017-06-29:
For DAB you need an VHF III antenna (the age does NOT matter). Forget those mini antennas attached to DVB-T(2) sticks. They are optimized for UHF.

HansVanEijsden, 2017-06-29:
Talking about antennae: I'm using this one for DAB+: http://www.dabtuners.nl/slechte-dab-ontvangst-huis-even-knutselen-dan (in Dutch, I think Google Translate can help you out). You only need a piece of coax, a connector and some tools. 😃 Great reception here.

gvanem, 2017-06-30:
@andimik They are optimized for UHF.

I admit, I do only have one such DVB-stick antenna. But it works fine for FM (which is VHF) with SDR-Sharp. I see a good spectrum in Welle-IO. With AGC off and a manual gain of 28dB, I get a lot of these warnings:
```
RTL_SDR: ADC overload. Maybe you are using a to high gain.
```
But the spectrum looks "lovely":
[image]

As far as I understand DAB, the data is split across those small QPSK carriers (to avoid *Multipath Fading*) and then recombined after demodulation. Isn't that so? I see a lot of:
[image]

toggling between "**OK**" and "**Not synched**. So maybe this is an overload problem? With AGC enabled, the situation is no better.

@andimik Do you have a DAB radio as well for testing the field strenght?

I do have this [3-mode radio](https://www.manualslib.com/manual/1092652/Radionette-Rndib13e.html) 3 meters away from the DVB-stick. It receives DAB+ with no problems.

andimik, 2017-06-30:
Your spectrum looks awful (for a city being covered with DAB like yours)

Use another antenna.

HansVanEijsden, 2017-06-30:
This is my very simple antenna as I mentioned above:

[image]

Here is the spectrum of channel 6B here:
[image]

And here channel 12C:
[image]

What you want is a good SNR, with low readings on the left and right of the spectrum, and high peaks in the middle.

andimik, 2017-06-30:
Please don't recommend such a self built antenna when your spectrum looks strange, too.

And don't pay lots for DAB sticks or antennas. 5-10 Euro should be the limit. People are throwing away these en masse. So just check second hand webpages.

gvanem, 2018-01-06:
Hi again. I've got a [better antenna](https://www.ebay.com/itm/391876292482) for my SDR-stick. 
Welle.io now works super nice. Including all (?) the sub-carriers showing the images:
[image]

PS. It seem enabling the Viterbi Spiral code had some effect. I notice it was disabled in the git-code. But I've compiled everything using MSVC-2017 with SSE2/AVX2. But Welle.IO sucks approx. 12% CPU constantly 🤒 .

AlbrechtL, 2018-01-06:
Great!

Regarding the Spiral code. It was disabled because the Spiral code needs a static length. But the most of the Viterbi decoding needs a variable length. To use the Spiral code for all the decoding was a bug where segmentation faults could occur.
The SSE optimization is only for the Spiral code which is disabled now. This is the reason that the CPU load doesn't change of you compile it for SSE.

gvanem, 2018-01-08:
>  To use the Spiral code for all the decoding was a bug where segmentation faults could occur.

I assume you mean with gcc. I can confirm it crashes with `clang-cl` (which is close to `gcc`). But if I do:
```
viterbi(int16_t, bool spiral = true);
```
using MSVC-2017, it works fine (no crash on *non-16-byte aligned* accesses). I assume you have not tried MSVC since the code only seems to support `__MINGW32__` on Windows (which IMHO is a big pity).

AlbrechtL, 2018-01-10:
Did you successfully compiled welle.io with MSVC?

The problem with the spiral code is that you access invalid parts of the RAM. You can analyze it with the tool [valgrind](http://valgrind.org/). I saw the most crashes under Android (https://github.com/AlbrechtL/welle.io/issues/123).

gvanem, 2018-01-12:
> Did you successfully compiled welle.io with MSVC?

Yes, but with lots of patches; mainly replacing `char var[variable_len]`; with `alloca()` code. MSVC-2017 is still not fully C-99.