AlbrechtL/welle.io #58: audio stops wile brouwsing internet

glis011, 2017-05-06:
de audio drops sound if i listen to a station and brouwsing the web

AlbrechtL, 2017-05-06:
welle.io needs a lot of CPU power.
Which CPU do you are using?
Can you report your CPU load please?

glis011, 2017-05-06:
AMD dual core 5000+ 4gb ram

AlbrechtL, 2017-05-06:
Your processor is 9 years old and has only two cores. Welle.io is optimized for 4 cores. That the sound stops while surfing is simply because your CPU is not powerful enough.

glis011, 2017-05-06:
Ok THX for the information but it works also on PC if i dont do someting else

tnadolecki, 2017-05-07:
I have Core i5 520M - 2.4 Ghz  with 4Gb RAM which I though was decent but I also need to shutdown all other programs while running welle.
Few seconds and cooling fan kicks in. It's not suitable for long listening unless you like overheating your gear and listening to screaming fan.

AlbrechtL, 2017-05-08:
@tnadolecki: Even your processor is 7 years old. For using welle.io without a fan please use a device with a modern CPU like Intel Atom Z8300 or Intel Atom Z3736F . Those CPUs can handle welle.io and are working with passive cooling.

closed 2017-05-08 by AlbrechtL