AlbrechtL/welle.io #59: Shader

Apklink, 2017-05-17:
I am using Windows 7.  I get a message "shader program not linked"  What should I do?

Apklinl

AlbrechtL, 2017-05-17:
Please post a screenshot of the command line window.
Which graphic card do you have?

Apklink, 2017-05-18:
Thank you for your quick response.
I am not very computer literate and some messages which may appear logical to many, do not make sense to me.  I hope that the attached screen shots give you all the information you need.
Feel free to reply in German  I speak and read German perfectly, but I cannot write.
Regards,
Chris
[image]
[image]
[image]
[image]
[image]

AlbrechtL, 2017-05-18:
Did you use the latest nvidia driver?
It is more likely a QT and OpenGL problem. You can also try to compile welle.io direct from the source https://github.com/AlbrechtL/welle.io#windows-10 and try it again.

Apklink, 2017-05-18:
Thank you Albrecht.
I will check for for the latest nividia driver.  I am afraid that with my knowledge of computer software "QT' and OpenGL" mean nothing.  Compiling welle,io is well beyond my capabilities and as I am getting close to 80 it is a bit late to start learning new things,
Thanks again for your help

AlbrechtL, 2017-05-21:
If the latest nvidia driver doesn't help please use Windows 10 (Windows 7 is officially not supported).

closed 2017-05-21 by AlbrechtL

Apklink, 2017-05-22:
Thank you again Albrecht.  
I am sorry that I have not commented earlier, I think that since you have taken the trouble to provide suggestions, common courtesy demans that I at least keep you updated on developments.  However it took me some time to install the latest driver for the Nividia Quadro FX1500 graphics card.  Windows could not manage the automatic update and only last night I managed to update the old driver (2009!) to the most recent one.
It now works like a charm!!
Hence I can confirm that the program works with Windows7 Home Premium and the Nividia Quadro FX1500 graphics card.  I am using the RTL-SDR2832 dongle.

Regards and thanks again, Chris