AlbrechtL/welle.io #61: Tuning time is very long

russel, 2017-05-18:
Doing a tuning over all the DAB channels takes a long time. DVB is similar. DVB have the idea of transmitter to limit the number of channels search for stations. This will not work the same for DB because of the way transmitters are handled. However countries do have limitations on which channels are used for DAB. If there was a notion of a country tuning file to restrict the tuning ranges, tuning would be a lot quicker. I believe commercial  DAB radios do something along these lines.

russel, 2017-05-18:
Caveat though: it is great that the tuning is cached between executions. However, what happens after a move of the computer and device to a different location, is the location checked?

AlbrechtL, 2017-05-19:
The issue is that you can't know the current location. A user could set the location but then you need files with the location specific channel information. At the moment these files don't exists. My experience with DVB-T in Germany is that those files are outdated all the time.

How much time does the channel scan need in your case?
I think a better approach is to speed up the current channel scan algorithm.

russel, 2017-05-19:
I guess I am used to working with Debian Sid and Fedora Rawhide where the UK transmitter DVB data gets updated very quickly. The DAB transmitter system is somewhat less coordinated! If there is no big effort to get DAB transmitter data in place, as has been done for DVB, there is no point. The question is whether to start it or not.

It took about 6 or 7 mins for the first tune. I guess I should remove the tuning data in the config file and try again properly timed. On the up side, this is only needed on moving the device.

AlbrechtL, 2017-05-20:
I personally don't have a deep interest to start such a project and to maintain it. If you would like to do it, go ahead! You are welcome to send pull requests for handling such files or databases in welle.io.

russel, 2017-05-21:
Understood.

closed 2017-05-21 by russel