AlbrechtL/welle.io #63: Does not find Pulse on Fedora Rawhide

russel, 2017-05-18:
I keep getting the problem:

    PulseAudioService: pa_context_connect() failed

and a lack of audio. Is this that the CMake build is failing to do something during the build?

russel, 2017-05-19:
Could be the same issue as #64 – different OSes different error reports.

AlbrechtL, 2017-05-19:
I suppose QT tries to use Pulse and Pulse somehow fails on Fedora. Does other multimedia software e.g. VLC is working?

russel, 2017-05-19:
The AppImage behaves a little differently.

My build shows the connect failed, and no matter how much tuning gives no further output on the console, but the tune seems good according to the UI.

The AppImage I downloaded behaves the same on startup, but then on a tune behaves as my build does on Debian. Tries to use Raw audio and fails.

```
PulseAudioService: pa_context_connect() failed
Audio: Raw audio format not supported by backend, cannot play audio.
RadioController: Number of gain steps: 28
```

AlbrechtL, 2017-05-20:
Are other audio output software is working e.g. VLC or Firefox with Youtube? I assume something is wrong in your pluse installation.

russel, 2017-05-21:
Sound is working entirely fine with Rhythmbox, Chrome, Firefox, etc. Thus PulseAudio appears to be working fine.

AlbrechtL, 2017-05-21:
Mh, maybe something goes wrong with the [default audio device selection](https://github.com/AlbrechtL/welle.io/blob/0e258bb87957a0ec4ecc4be412529ac203fba488/src/output/CAudio.cpp#L76). You can try just to comment out line 76 - 81.

russel, 2017-05-21:
This may be an Heisenbug :-( I just tried running it on Fedora and it works fine. This is bizarre. But good.

russel, 2017-05-22:
Closing this for now. Not tried on Debian recently. If it has a problem I'll reopen or create a new issue.

russel, 2017-05-22:
@AlbrechtL This does now seem working normally on Fedora. I cannot explain why it failed before.

russel, 2017-05-22:
A definite Heisenbug here. Welle.IO worked this morning. Now when I start it QtAudioPulse registers with the Fedora sound system, but when I tune, I now get:
```
r82xx_write: i2c wr failed=-4 reg=05 len=1
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
r82xx_write: i2c wr failed=-4 reg=05 len=1
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
r82xx_write: i2c wr failed=-4 reg=05 len=1
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
r82xx_write: i2c wr failed=-4 reg=05 len=1
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
r82xx_write: i2c wr failed=-4 reg=05 len=1
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
r82xx_write: i2c wr failed=-4 reg=05 len=1
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
r82xx_write: i2c wr failed=-4 reg=17 len=1
r82xx_set_freq: failed=-4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
```

russel, 2017-05-22:
And now it is working fine on Fedora Rawhide. I am assuming there is a state of the hardware that is being reached that Welle.IO is not able to reset.

OK, I touched the NooElec unit and the sound stopped and Welle.IO reports an error from the device. dmesg not reporting any change of state. Restart Welle.IO and everything is fine. This is getting quite annoying. Clearly something is right, but there are too many instabilities.

Or maybe I am just getting over hyper. :-)

AlbrechtL, 2017-06-21:
I assume there is an electrical problem in your NooElec device in combination with a rtlsdr driver issue. Anyway, I close this issue for now and if you have more information we can reopen it.

closed 2017-06-21 by AlbrechtL