AlbrechtL/welle.io #64: Cannot play raw audio on Debian

russel, 2017-05-19:
Having build Welle.IO on Debian using the CMake build, then on execution and tuning to a station:

    Audio: Sample rate 48000 Hz
    Audio: Raw audio format not supported by backend, cannot play audio.

This is apparently a different problem to the Fedora one (Issue #63) but may in fact have the same basic cause.

AlbrechtL, 2017-05-19:
Interesting. This error comes from [CAudio.cpp](https://github.com/AlbrechtL/welle.io/blob/0dd9bcb4493b4ae7198db0a8c05c345329b7c2ec/src/output/CAudio.cpp#L77). 

Can you try the [AppImage](https://github.com/AlbrechtL/welle.io/releases/download/V20170417/20170417_Linux_welle-io.AppImage) please, just to verify if it is a QT related problem.

russel, 2017-05-19:
The AppImage appears to work fine. So it must be something in my build. :-(

djnutmac, 2017-05-19:
I tried the appimage with chmod a+x xxxx.AppImage
but the program does not start

AlbrechtL, 2017-05-20:
@djnutmac Please open a new issue with an error message or screen shot of the problem.

AlbrechtL, 2017-05-21:
Closed because it seems to be duplicated with #63.

russel, 2017-05-22:
@AlbrechtL It seems that even though the build works on Fedora Rawhide now, it still fails on Debian Sid, so this issue needs reopening. :-(

russel, 2017-05-22:
I also notice that there is a lot more logging information output when the application is running on Debian Sid compared to Fedora Rawhide and yet the builds should have had exactly the same parameters set.

russel, 2017-05-22:
Tried commenting out the lines in CAudio.cpp and:
```
RTL_SDR: Increase gain to 48
RTL_SDR: Increase gain to 49.6
DABConstants: Unknown language type
dab-audio: we have now DAB
RadioController: Audio sample rate 48000 kHz
Audio: Sample rate 48000 Hz
using null output device, none available
using null output device, none available
RTL_SDR: Decrease gain to 48
RTL_SDR: Increase gain to 49.6
RTL_SDR: Decrease gain to 48
```
so the code is needed.

russel, 2017-05-22:
It is clear now that on Debian the Qt sound system is not registering properly with the OS sound system: on Fedora, starting Welle.IO causes the OS sound system to receive registration of a QtPulseAudio device, and the sound works. This registration does not happen on Debian Sid, and no sound.

AlbrechtL, 2017-05-22:
Which QT version do you use on Debian?

russel, 2017-05-22:
Debian Sid is currently using Qt 5.7.1. Fedora Rawhide is on 5.9.0-beta-3.

AlbrechtL, 2017-05-22:
OK, these versions are fine. At the moment I don't have any ideas what is the problem. You are the first user that reports such errors. Do you have a special sound card?

russel, 2017-05-27:
Tried on another computer, so different sound card. Exactly the same behaviour: my build appears to work fine but no sound (the raw sound message), the AppImage appears to work fine. So this seems to be a "Russel's Debian Build" thing. The problem is I am not sure how to investigate this, since the build actually works fine: I am at a bit of a loss how to proceed.

AlbrechtL, 2017-05-28:
Are you using the standard QT Debian packages or the binaries from the official QT website? Maybe the QT Debian packages are built with different compiler switches compared to the official binaries.

russel, 2017-05-29:
I get CMake to determine the presence of the dependencies using the Qt CMake additions. So on Fedora it will be Fedora packaging, on Debian it will be Debian packaging. In the next few days, I'll build for Fedora and then for Debian on the same machine will full tracing logs, and see if I can see where the Pulse registration is failing for Debian. I had been hoping to avoid this sledgehammer approach, but it seems it will have to be done.

AlbrechtL, 2017-05-31:
FYI: In a0a1d9d I changed the error message to *Audio format "audio/pcm" 16-bit stereo not supported. Something went wrong, cannot play audio.* This doesn't improve anything but it gives a little bit more clear message.

russel, 2017-06-10:
OK we have a resolution. Nothing wrong with the build – which I believed as it all worked on Fedora Rawhide. The problem is the different way Debian packages compared to Fedora. So for Debian you have to install the multimedia plugins package separately. In particular, libqt5multimedia5-plugins has to be installed manually. It now all works fine on Debian, thus I am closing this issue.

closed 2017-06-10 by russel