AlbrechtL/welle.io #67: Floating text for Credits (under Help)

andimik, 2017-05-20:
It makes sense to have the text for the credits as floating text as there is an automatic word wrap, indeed.

AlbrechtL, 2017-05-21:
You can do this and send me a pull request.  

You can find the text in [AUTHORS](https://github.com/AlbrechtL/welle.io/blob/0e258bb87957a0ec4ecc4be412529ac203fba488/AUTHORS) and [COPYING](https://github.com/AlbrechtL/welle.io/blob/0e258bb87957a0ec4ecc4be412529ac203fba488/COPYING). To enable or disable word wrap play with [line 20 in InfoPage.qml](https://github.com/AlbrechtL/welle.io/blob/0e258bb87957a0ec4ecc4be412529ac203fba488/src/gui/QML/InfoPage.qml#L20).

Official QT documentation for wrap mode: https://doc.qt.io/qt-5/qml-qtquick-text.html#wrapMode-prop.

andimik, 2017-06-11:
See Pull request #84

closed 2017-06-12 by AlbrechtL