AlbrechtL/welle.io #70: Font and font size

russel, 2017-05-22:
When using Welle.IO on a workstation the fonts are too big. Also they are the wrong fonts for the theme.

Clearly it cannot be expected for a Qt application to obey a GTK/GNOMEShell theme, but it would be good to be able to change the font explicitly in the Qt application. This is possible in other Qt realised applications I am using. The question is whether QML allows the flexibility Qt does with fonts.

AlbrechtL, 2017-05-22:
The font size is automatically calculated depending on your screen DPI to support high DPI displays. So it is not just to set the font size. I also think that the most of the users don't care about the font size and they don't want to change it.

You are welcome to update the font size if you like e.g. [here](https://github.com/AlbrechtL/welle.io/blob/0e258bb87957a0ec4ecc4be412529ac203fba488/src/gui/QML/style/Style.qml). You can also propose something to adapt the font size better to the users OS. On Android for example the font size is a little bit to small. So there is potential to make it better.

AlbrechtL, 2017-06-27:
I'm closing this issue because there is no discussion about this right now.  

If somebody would like to discuss this topic they can use this issue.

closed 2017-06-27 by AlbrechtL