AlbrechtL/welle.io #71: Does not see newly added device

russel, 2017-05-22:
If Welle.IO is started with no RTLSDR device present it reports it missing. If a device is then added, this is not recognised on an attempt to retune, which goes into a loop (presumably a timeout of some sort):
```
…
RadioController: Start or restart device
RadioController: Radio device is not ready or does not exits.
RadioController: Tune to channel "11B" -> 218.64 MHz
OFDM-processor: closing down
OFDM-processor: start
RadioController: Tune to station "Absolute C Rock"
RadioController: Restart syncing. isSync: false , isFICCRC: false , FrameErrors: 0
RadioController: Tune to channel "11B" -> 218.64 MHz
OFDM-processor: closing down
OFDM-processor: start
RadioController: Tune to station "Absolute C Rock"
RadioController: Restart syncing. isSync: false , isFICCRC: false , FrameErrors: 0
RadioController: Tune to channel "11B" -> 218.64 MHz
OFDM-processor: closing down
OFDM-processor: start
RadioController: Tune to station "Absolute C Rock"
…
```

AlbrechtL, 2017-05-22:
Yes, the behavior is correct. welle.io only detects devices while starting but not dynamically during runtime.
I added this feature to the [task list](https://github.com/AlbrechtL/welle.io/wiki/Open-Tasks)

closed 2017-05-22 by AlbrechtL

russel, 2017-05-22:
For Me TV I have used inotify for dynamically attaching inserted USB devices, but this may be too Linux specific to be useful.

viktorgino, 2017-05-22:
libusb has a [hotplug API](http://libusb.sourceforge.net/api-1.0/hotplug.html), just need to setup an event handler loop for it and start/stop DAB on demand.

AlbrechtL, 2017-05-22:
@viktorgino Great idea! But to use it you have to enhance [librtlsdr ](https://github.com/steve-m/librtlsdr) which uses libusb. welle.io just uses librtlsdr.

viktorgino, 2017-05-23:
Only recommended it because I remembered it being a dependency. On linux you could also use libudev, but its not even remotely cross platform.