AlbrechtL/welle.io #73: Missing main window and expert window under Android

andimik, 2017-05-23:
Today not even the main window and the expert window dissappeared (was grey), the exit button also was not visible. I had to rotate the mobile and then they appeared again.

[image]

AlbrechtL, 2017-05-23:
Which mobile phone and Android version do you are using?

andimik, 2017-05-23:
Sony Xperia Z1 Compact
Android 5.1.1 (latest version for this mobile) 

I tried to find out what caused the problem and then maybe this has to do with the graphics card 

The grey area right has the same size as the Home, back and switch buttons.

[image]

andimik, 2017-05-23:
[image]
At the bottom you see this grey area, same size as in the upper screenshot.

AlbrechtL, 2017-05-23:
OK, now I understand your problem. This should be fixed in the next version. I already committed the fix (43e76be).

closed 2017-05-23 by AlbrechtL