AlbrechtL/welle.io #74: RTL_SDR: continuous increase and decrease of gain

lucaberta, 2017-05-23:
Hi @AlbrechtL I have just discovered your great project and have installed Qt 5.7 and successfully run the latest Git code on my Ubuntu 16.04 machine, using an R820T USB tuner.

Here in the Lausanne area there are 3 multiplexers, and while with 2 of them the log window shows the "RadioController: Found station" lines followed by a "dab-audio: we have now DAB+" log line, on the 3rd multiplex the log window keeps on showing alternating "Increase gain" and "Decrease gain" while at the same time the tuning and audio decoding seems to be working perfectly.

Let me know if I can provide any more debugging information, and which ones might be useful.

Many thanks, Luca HB9FGJ

AlbrechtL, 2017-05-24:
Your discovered behavior is correct. It is the automatic gain. One value is to small and the other value is to big thus it is alternating.

closed 2017-05-24 by AlbrechtL

lucaberta, 2017-05-24:
Thanks for your comment @AlbrechtL , I am happy to see that I have done nothing wrong given that I am a first timer!

Still, would it not make more sense to have a wider margin for hysteresis to avoid this flip-flopping which makes the debug window unreadable?

It cannot be possible that there is NO right gain level to set, seems crazy to me.

Surely I can turn off automatic gain and go manually, but since there are 3 different multiplexers in the area, different gains are needed.

Thanks, Luca

AlbrechtL, 2017-05-24:
I definitely agree that a hysteresis would be useful. You are welcome to send me a pull request. In my location I only work with the maximum gain because the signal is so weak. So the automatic gain is not fully tested.

AlbrechtL, 2018-01-29:
I added it to the [task list](https://github.com/AlbrechtL/welle.io/wiki/Open-Tasks).

mpbraendli, 2018-05-07:
There is some hysteresis indirectly because the criteria to increase gain is not the same as to decrease gain. We could replace the `NewMaxValue <= 255` condition by something a bit a bit lower than 255 to increase the hysteresis.
@lucaberta can you try some values in `src/input/CRTL_SDR.cpp` (assuming you use rtl_sdr and not rtl_tcp) and tell us if it is still flapping?