AlbrechtL/welle.io #75: Gain index out of range

LarsHaack, 2017-05-24:
I tested welle.io while driving from home to work using a laptop with Win10.
The AGC is enabled as I was basically interested in the behavior of the AGC based on IQ data.
I know that I need the full range of gain levels of my NooElec NESDR Stick with R820T tuner for this route.
In an area with high signal level I get below debug log:

...
RTL_SDR: Decrease gain to 0
RTL_SDR: Increase gain to 0.9
RTL_SDR: Increase gain to 1.4
RTL_SDR: Decrease gain to 0.9
RTL_SDR: Decrease gain to 0
RTL_SDR: Decrease gain to -1.94616e+08
RTL_SDR: Unknown gain count 65534
RTL_SDR: Decrease gain to -1.94616e+08
RTL_SDR: Unknown gain count 65534
...

It seems to me that the gain index currently being 0 shall be decreased agin in CRTL_SDR::AGCTimerTimeout() under these conditions.

AlbrechtL, 2017-05-24:
Thanks for reporting this issue.
In my location the signal is very weak so the AGC is only tested with weak signals :-(. Good that you tested it with a really strong signal.
If you are a developer please fix this issue and send me a pull request.

AlbrechtL, 2017-05-31:
Fixed in d67d07d. Will be included in the next release.

closed 2017-05-31 by AlbrechtL