AlbrechtL/welle.io #76: Can't find stations on Raspberry Pi/Raspbian

viktorgino, 2017-06-01:
When running on the Pi can't find channels, when I add the channels manually to the config file it can't tune to them. I use Rasbpian Stretch Lite with the latest kernel (4.9.30-v7+). I tried it with the same Realtek stick and antenna as I use on my desktop and I get perfect signals with it.

I'm out of ideas on what could be causing this, if anyone has any idea on how to go about debugging this then let me know.

AlbrechtL, 2017-06-01:
Please post the command line output.

viktorgino, 2017-06-01:
[Console output on the Raspberry Pi](https://github.com/AlbrechtL/welle.io/files/1044054/welleio-rpi.txt)
And for reference, trying to tune the same station with the same setup: [Console output on Ubuntu Desktop](https://github.com/AlbrechtL/welle.io/files/1044053/welleio-ubuntu.txt)

AlbrechtL, 2017-06-01:
The logs seems to be OK. But you likely run into a CPU overload. Did you compile welle.io as release (with compiler optimizations) or debug (without compiler optimizations)? welle.io runs only with compiler optimizations on the RPI.

viktorgino, 2017-06-01:
Managed to miss that bit, even tho I've seen it in the project file 😃. Thanks! Works on the Pi pretty good now!

AlbrechtL, 2017-06-01:
Great!

closed 2017-06-01 by AlbrechtL

AlbrechtL, 2017-06-01:
I just saw that you developed an alternative GUI for welle.io for your [headunit-desktop](https://github.com/viktorgino/headunit-desktop) project. I invite you to contribute code to welle.io.

viktorgino, 2017-06-01:
Will do, I'll put together a pull request for it.