AlbrechtL/welle.io #77: Can't interact with the content of SplitView

viktorgino, 2017-06-01:
This is quiet possible a Qt bug, but here we go:

When using a touch screen (native touch screen, without X and mouse emulation) I can't interact with the controls of the GUI that are within a SplitView, I can only drag the separator. When trying to use a mouse the same thing happens, plus the mouse pointer is "horizontal drag". I've tested this using a Raspberry Pi + the Pi Foundation's official 7inch LCD.

I'll do a quick proof-of-concept to confirm whether this is a Qt related issue, but in case this is Qt related, would you be open to an alternative for the SplitView based layout?

viktorgino, 2017-06-01:
I believe this is Qt related as I could reproduce the issue outside of this project. See bug report : https://bugreports.qt.io/browse/QTBUG-61148

AlbrechtL, 2017-06-01:
You may can try QT 5.9.

viktorgino, 2017-06-01:
Might do, but will have to update Raspbian to experimental first

viktorgino, 2017-06-01:
To sort this I have to launch welle.io with `QT_QPA_EGLFS_PHYSICAL_WIDTH=154 QT_QPA_EGLFS_PHYSICAL_HEIGHT=86` otherwise it can't calculate the screen's pixel density, which is weird as it should default to 100DPI. This can be closed.

closed 2017-06-01 by AlbrechtL