AlbrechtL/welle.io #78: Adding the 3D view from headunit-desktop

viktorgino, 2017-06-01:
[image]
[image]

AlbrechtL, 2017-06-01:
This looks great! A totally different approach to select a station. It could be a fantastic addition to the existing once.
Inside the colored rectangle we can also put the station logo. But for this a small picture database is necessary.

viktorgino, 2017-06-01:
Haven't implemented it yet, but I want to load the MOT slideshow in the current channels rectangle. As for the logo, there could be a folder where the user can put the logos and the app can look for the file looking its station name.

viktorgino, 2017-06-01:
Would you like this GUI optional or have it as the only GUI?

AlbrechtL, 2017-06-01:
> [...] but I want to load the MOT slideshow in the current channels rectangle

Also a great idea.

> Would you like this GUI optional or have it as the only GUI?

Difficult question. For now I think this could be optional. But lets see... If your 3D GUI is more sophisticated than the existing once why not to switch? To maintain one GUI is better then to maintain two GUIs.

viktorgino, 2017-06-04:
Please see: https://github.com/AlbrechtL/welle.io/pull/80
What I haven't integrated yet is the MOT slideshow and the radio station art

AlbrechtL, 2017-06-30:
I closed it because the 3D GUI is not ready yet. But we can use this issue to discuss the nice 3D GUI.

closed 2017-06-30 by AlbrechtL

viktorgino, 2017-07-03:
Created a new [pull request](https://github.com/AlbrechtL/welle.io/pull/99).

I might go ahead and make one settings page (I rewrote it for the 3D layout), just to remove redundancies and make code maintenance easier.