AlbrechtL/welle.io #81: Android: HE-AAC v2 stations shown as Mono

es5nhc, 2017-06-06:
Greetings. I am writing to bring your attention to something that I just observed on at least the Android version of welle.io. During a strong tropospheric ducting this morning I was able to pick up Latvian DAB+ test on my SDR dongle and I noticed that the services which are transmitted at 64 kbps with HE-AAC v2 were marked as Mono. Meanwhile, my Technisat Techniradio 1 as well as Wavesink app both said the channels were SBR+Parametric Stereo(indicating HE-AAC v2). Attached is a screenshot from the Android app. The version on the phone is 1.0-alpha, Build on: Sat May 20 2017 14:49:07. 
[image]

If you have any further questions, feel free to let me know

AlbrechtL, 2017-06-06:
If possible please record a [RAW file](https://www.welle.io/devices/rawfile#record-an-u8-raw-file) and provide it to use.

es5nhc, 2017-06-06:
I will try to record an I/Q dump with SDRTouch (hard disk on the computer is too slow for anything good longer than a few seconds) next time there is a tropo opening. This might take time as I am 195 km from the transmitter and therefore have no regular reception.

viktorgino, 2017-06-06:
@es5nhc How big of an antenna are you using? You can pick up some nice directional antennas online that should help you get a better reception.

es5nhc, 2017-06-06:
In this case I was using my 4 element DK7ZB for 2m band. The multiplex is also only transmitting at 1.5 kW AFAIK.

nottledim, 2017-08-14:
BBC3 (classical music) always shows mono too.  But qt-dab has same problem.

andimik, 2017-08-15:
But this is old DAB. It is 192 kBit mp2 at PL 3

Can be tested via Satellite. http://www.fmlist.org/sendertabelle/dab-ww_more.php?inputLand=G&inputEn=CE15 and the stream is I guess also at http://tipok.org.ua/node/46

AlbrechtL, 2017-08-15:
Can you provide a [RAW file](https://www.welle.io/devices/rawfile#record-an-raw-file) please?

es5nhc, 2017-08-25:
Re: UK issue, I can confirm. I just found it on a WAV file that I recorded in the UK last month. Additionally I was unable to decode DLS while there, including on this multiplex.

https://drive.google.com/file/d/0Bz0aEk5iHH62T0VVYTBMeXRlck0/view?usp=sharing

Probably should be moved into its own issue.

Off topic note: If you use SDRTouch on to record raw files, ensure that the center frequency was in the dead center of the multiplex in order for it to play successfully.

AlbrechtL, 2017-08-25:
Thanks for the RAW-file.
In 18e9a6d I reintroduced the PAD decoding for DAB including DLS.

[image]

The mono/stereo issue is still unsolved for both DAB and DAB+.

es5nhc, 2017-09-23:
Regarding Latvia: Thanks to a megatropo that has just begun in the Baltics, I can finally contribute an I/Q recording. I can also confirm the issue outlined in the initial post still exists. It isn't perfect as the signal strength changes(probably a good test case for weak signal reception ;) ) but should be sufficient to replicate what I am experiencing. The affected channels are LR5 Atklājumi, LR5 Hiti, LR5 Latvieši, LR5 Rīti. Listening to the audio you can clearly hear stereo effects.

Here's the recording: https://drive.google.com/file/d/0Bz0aEk5iHH62VkFFYTZ2RDladTg/view?usp=sharing

AlbrechtL, 2017-09-23:
Thanks for the file. It helps a lot to find the problem.

AlbrechtL, 2018-01-28:
I found the issue.

It was not really a bug. [Parametric stereo](https://en.wikipedia.org/wiki/Parametric_Stereo) uses a mono audio signal and add some additional information to reconstruct the stereo signal. So the mono indication was actually correct.
In e8fe11a2d70c610ccb5ecb71133447767eccc992 I added some additional code to detect the parametric stereo signal and show the signal as stereo in the GUI.
[image]

@JvanKatwijk If you like you can fix the same issue in your software.

closed 2018-01-28 by AlbrechtL