AlbrechtL/welle.io #85: Radiocontroller: Restart syncing. isSync: true. isFICCRC false. FrameErrors 25

gtekprog, 2017-06-12:
Hello Gents,

I receive this error ( with a complete ofdm resync ) after a fixed period. Always 25 frameErrors.

What is wrong?

BR

AlbrechtL, 2017-06-13:
Please tell us your location (country), channel (frequency) and station. Does it happen only with a specific station in one channel or with all stations?  

If possible record a [RAW file](https://www.welle.io/devices/rawfile#record-an-u8-raw-file) and provide it to use.

gtekprog, 2017-06-13:
Hi Albrecht,

IT is in the Netherlands . Channel 5D.
The station uses odr software ( IT is ours )

Br.

Evert

AlbrechtL, 2017-06-13:
Please provide us a [I/Q RAW file](https://www.welle.io/devices/rawfile#record-an-u8-raw-file). This file contains the modulated signal as it is broadcasted. With this file we can reproduce the bug.

gtekprog, 2017-06-13:
Hi Albrecht,
Will send you the RAW file tonight. It could as well be that our MUX is producing skips. However, this is not heard, seen on other DAB+ radio's.
Evert
----Origineel Bericht----

andimik, 2017-06-14:
Off topic: see http://wohnort.org/dab/ where Carey reports double IDs on different ensembles... Shall be fixed in order to guarantee better handling on user side.

AlbrechtL, 2017-07-25:
Closed because there are no more information to reproduce the problem.

closed 2017-07-25 by AlbrechtL