AlbrechtL/welle.io #86: Automatic gain still inferior to manual setting

mickdick, 2017-06-18:
The automatic gain improvement in V1.0 2017-06-17 helped somewhat (also in station scanning) but for reception the manual setting still delivers much better results (in terms of SNR and disappearance of reception drop-outs).

AlbrechtL, 2017-06-18:
Which OS and dongle do you use? Please give more details what better results means?

mickdick, 2017-06-20:
Hello, 

 

the OS is Android 5.1 running on a JayTech TPC XTE10D Tablet (cheap 

stuff but working fine also with OTG). The dongle is from a company

called "Sodial" featuring the R820T + RTL2832U Chips.

 

In the one session I found consistently more drop-outs with the new

automatic gain setting than with a manual gain value (around 33 I think).

However, I found it hard to reproduce the effect yesterday and have to 

go back to more systematic testing.

 

It is ok for me if you cancel this entry. I will re-submit with more detailed

information if I can reproduce the effect more clearly. 

 

Thanks for the very good implementation reached already!

 

Regards,  Michael

AlbrechtL, 2017-06-21:
OK, I close it.
Please leave a message here if you have more information then I will open it again.

closed 2017-06-21 by AlbrechtL