AlbrechtL/welle.io #88: [Android] Playing stops if app not in foreground

AlbrechtL, 2017-06-22:
Under Android welle.io stops playing if app is not in foreground.

I invite everyone to write into this issue if it is missing.

twoofseven, 2017-07-20:
Will be solved with #103

joachimkr, 2017-09-06:
Would be great!
Any Standby currently interrupts operation.
If I close the cover on my Galaxy A5, the display is switched off. Similar to standby.
Welle.io does not tune anymore, after reactivation. May be driver related too, but Martin's sdr app survives the cover close.
Greetings,
Joachim

AlbrechtL, 2017-09-06:
You can download a test APK [here](https://gist.github.com/AlbrechtL/1fb6cc890863b21fe524b9e9ea9766d0/raw/4308fa90d5827254c6932f8b61f4bd7e6e591b95/20170905_Android_armv7_welle-io.apk).

Please report if background playing works now.

This version includes also a first version of an Android Auto integration. For more information please visit #103.

joachimkr, 2017-09-08:
Hello Albrecht,

It is working very well!
Galaxy A5(2016).
I can continue listening while writing email 😎

Reception quality also seems better (changed something?)

Best Regards
Joachim

joachimkr, 2017-09-08:
Just a minor issue: Stopping via notification area would be great.
I just unplugged.

twoofseven, 2017-09-09:
Thanks for your proposal.
I already implemented the Notification’s setShowCancelButton functions, because I had the same issue. Unfortunately this seem to be only shown on Android <5.0.0 (according to certain documentation).
I was not sure, whether a deducated stop button direcly next to the playback control buttons would make sense.
Another idea could be to not lock the notification and close the background service, if the notification got closed/deleted (swiped away).

joachimkr, 2017-09-09:
I use a Android 7 on  my Galaxy. I can swipe the notification away, when it
is inactive. But I need to pull the USB component, to stop sound/receiving.
Greetings, Joachim

AlbrechtL, 2017-09-17:
Thank to twoofseven there is an improved test version available: [20170917_Android_armv7_welle-io.apk](https://gist.github.com/AlbrechtL/1fb6cc890863b21fe524b9e9ea9766d0/raw/44974c20fcf182e9552fb021a6a98326c9403810/20170917_Android_armv7_welle-io.apk)

Please try it again and report your results.

joachimkr, 2017-09-17:
Is there any native code for arm7 within this apk?
I would like to run welle.io on my car stereo, which is an Intel CPU / quadcore with Android 6. 
I now understood a bit about the hardware aspects. But I have no experience with programming Android applications. I am a Java programmer on SE and EE platform. I also have a history in C++. But not with Linux kernel.

I know, the RTL USB driver would be the companion part. Where can I find a starting point with that?

Greetings 
Joachim

AlbrechtL, 2017-09-17:
Yes, this test version is only for ARMv7.
But you find an official x86 version at the download page https://github.com/AlbrechtL/welle.io/releases.

twoofseven, 2017-09-18:
@joachimkr  On Android you do not need to work on Linux kernel level, since the Android RTL-SDR driver is an app (background service) connected via TCP-IP, the App is mainly written in C++ using Qt/QML and Java on Android Media Integration level.
I recommend to start with the Qt Open Source version including Qt Creator  from https://www1.qt.io/download/  and Qt Android support (http://doc.qt.io/qt-5/androidgs.html). I also installed the Android Studio to more conveniently develop on Android (java) and download the Android SDK /NDK.

joachimkr, 2017-09-20:
Hello,
I made a new attempt with the x86 apk.

It is running basically running now!
Not stable on application welle.io, but I have heard 4 seconds radio. I saw
the spectrum display.

So it seems, the hardware is capable, but antenna & receiver & Android
platform have noise or supply noise issues.

The channel scan is quite slow. The application crashes after some time, I
cannot reproduce the reason.
Most time, antenna level is low and the application is tuning...

I will try to get a noise reducing cable and restart attempts. I know, my
receiver depends on good conditions.

Can you give me a hint, how I can get detail about the app abortion?

Greetings,
Joachim

joachimkr, 2017-09-20:
Again, its me.
What happens, if the Android device CPU is too slow, what are the symptoms?

The response to user interaction is responsive, but some lazy behaviour can
be observed.
Greetings
Joachim

Am 20.09.2017 13:51 schrieb "Joachim Kröger" <
joachim.kroeger.pocketbiz@gmail.com>:

Hello,
I made a new attempt with the x86 apk.

It is running basically running now!
Not stable on application welle.io, but I have heard 4 seconds radio. I saw
the spectrum display.

So it seems, the hardware is capable, but antenna & receiver & Android
platform have noise or supply noise issues.

The channel scan is quite slow. The application crashes after some time, I
cannot reproduce the reason.
Most time, antenna level is low and the application is tuning...

I will try to get a noise reducing cable and restart attempts. I know, my
receiver depends on good conditions.

Can you give me a hint, how I can get detail about the app abortion?

Greetings,
Joachim

AlbrechtL, 2017-09-20:
Which CPU does the device have?

The discussion is drifting away from the origin thus I recommend to use the [forum](https://forum.welle.io) to discuss this topic.

joachimkr, 2017-09-20:
Ok. I entered there, do not get confirmation mail.

Here attached is a picture.

joachimkr, 2017-09-20:
got it. Entered Schaukasten.

AlbrechtL, 2017-09-20:
We continue the [car radio discussion](https://forum.welle.io/discussion/11/autoradio) in German inside the forum.

AlbrechtL, 2017-11-24:
Are added in https://github.com/AlbrechtL/welle.io/issues/103.

closed 2017-11-24 by AlbrechtL

joachimkr, 2018-01-06:
Hallo,
ich hoffe das ist in Ordnung in Deutsch zu schreiben...

Obiger Chip von NXP, kann welle.io damit umgehen?

Ich habe viel Zeit damit verbracht, auf einem x86-Android DAB in Gang zu
kriegen, hab es letztlich aufgegeben. Derselbe Empfänger konnte am PC
locker 28 Sender stabil empfangen, aber am Autoradio x86 ging fast nix.

Obiger Chip wird in einem neueren Radio mit Cortex A9 1.6Ghz Quadcore
angeboten. Aber die haben wohl selbst noch keine DAB Funktionen eingebaut.

Jetzt frage ich mich, ob ich das kaufe und Euch zur Verfügung stelle?

Viele Grüße aus Hamburg
Joachim Kröger

<https://gist.github.com/AlbrechtL/1fb6cc890863b21fe524b9e9ea9766d0/raw/4308fa90d5827254c6932f8b61f4bd7e6e591b95/20170905_Android_armv7_welle-io.apk>

AlbrechtL, 2018-01-07:
Welchen NXP Chip meinst Du?

joachimkr, 2018-01-09:
Hallo Albrecht,
im Betreff meiner Email war der exakte Chip genannt. Den hatte ein Anbieter
auf deinem Datenblatt genannt.
Der 'blöde' Client für gmail auf meinem Telefon gibt mir das nicht mehr
wieder...
Hier im Betreff ist eine Familie ähnlicher Chips genannt,  SAF360x, die nun
auch in Android Autoradios auftauchen.

Frage war, ob Ihr diese Empfänger unterstützt, wenn ja, wie ich das nutzen
kann, denn der Treiber für Realtek Chips kennt die natürlich nicht.

Gruß und Dank
Joachim

AlbrechtL, 2018-01-10:
I answer in English. The questions was if welle.io supports the [NXP chip SAF360x](SAF360x) which is build in inside car radios.

Unfortunately this chip doesn't have an I/Q-data interface. welle.io is an full SDR receiver and needs basically the RAW ADC data. On the other hand the SAF360x includes the full receiver and outputs the decoded audio samples. An application just configures this chip.

So the answer is that there is no support.

joachimkr, 2018-01-10:
Thanks for the clear explaination!
Regards
Joachim