AlbrechtL/welle.io #90: Appimage failing to load with Qt 5.7 not found error

asbaklm, 2017-06-24:
Hi there, I've tried running the welle appimages but am running into the following issue:

/usr/lib/x86_64-linux-gnu/libQt5Core.so.5: version `Qt_5.7' not found (required by ./welle-io.AppImage)

AlbrechtL, 2017-06-25:
Which Linux distribution and welle.io AppImage do you use? Please post the full command line output.

asbaklm, 2017-06-30:
Linux Mint 18.

> ./welle-io.AppImage 

./welle-io.AppImage: /usr/lib/x86_64-linux-gnu/libQt5Core.so.5: version `Qt_5.7' not found (required by ./welle-io.AppImage)
./welle-io.AppImage: /usr/lib/x86_64-linux-gnu/libQt5Core.so.5: version `Qt_5.7' not found (required by /tmp/.mount_U9ULwQ/usr/bin/../lib/libQt5Charts.so.5)

probonopd, 2017-07-01:
Hi @asbaklm please try to run on e.g., a Linux Mint 18 Live ISO. Does the issue persist?

asbaklm, 2017-07-08:
I tried Welle with an older version of Linux, Mint 17.3, and that is loading. One other difference being that I don't think that Qt is installed locally on that system.

When I try to start it on my main system, it still fails with:
./20170701_Linux_welle-io-x86_64.AppImage: /usr/lib/x86_64-linux-gnu/libQt5Core.so.5: version `Qt_5.9' not found (required by /tmp/.mount_201707Ro2iQK/usr/bin/../lib/libQt5Charts.so.5)

What could be happening is that if Qt is already installed, Welle may be calling the locally installed Qt libraries instead of using the ones inside the appimage. And because my local version of Qt is an earlier release it doesn't find the version of libQt5Core.so.5 that it is looking for and fails to execute.

So I'm wondering, can the appimage be built in such a way that it doesn't attempt to make use of locally installed Qt libraries, because that would be one of the reasons to use appimages in the first place instead of trying to compile apps to run on systems where locally installed libraries may not necessarily match the exact requirements of the app.

probonopd, 2017-07-08:
> So I'm wondering, can the appimage be built in such a way that it doesn't attempt to make use of locally installed Qt libraries

That's how it _should_ behave. Something is going wrong here...

probonopd, 2017-07-08:
Please run `LD_DEBUG=libs /home/linux/Downloads/20170701_Linux_welle-io-x86_64.AppImage 2>&1 | grep libQt5Core` and paste the output here. On my system it looks like this:

```
linux@linux:~> LD_DEBUG=libs /home/linux/Downloads/20170701_Linux_welle-io-x86_64.AppImage 2>&1 | grep libQt5Core
      8054:     find library=libQt5Core.so.5 [0]; searching
      8054:       trying file=/tmp/.mount_201707XrOodO/usr/bin/../lib/libQt5Core.so.5
      8054:     calling init: /tmp/.mount_201707XrOodO/usr/bin/../lib/libQt5Core.so.5
```

This confirms that the `libQt5Core.so.5` from inside the AppImage is being loaded, despite Qt being installed on my system too.

asbaklm, 2017-07-10:
I couldn't get it to display any output by grepping for libQt5core, so I made the search more generic. Here is the result.

LD_DEBUG=libs 20170701_Linux_welle-io-x86_64.AppImage 2>&1 | grep -i qt

      3216:	 search path=/opt/Qt/5.5/gcc_64/lib/tls/x86_64:/opt/Qt/5.5/gcc_64/lib/tls:/opt/Qt/5.5/gcc_64/lib/x86_64:/opt/Qt/5.5/gcc_64/lib:/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/tls/x86_64:/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/tls:/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/x86_64:/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms		(LD_LIBRARY_PATH)
      3216:	  trying file=/opt/Qt/5.5/gcc_64/lib/tls/x86_64/libpthread.so.0
      3216:	  trying file=/opt/Qt/5.5/gcc_64/lib/tls/libpthread.so.0
      3216:	  trying file=/opt/Qt/5.5/gcc_64/lib/x86_64/libpthread.so.0
      3216:	  trying file=/opt/Qt/5.5/gcc_64/lib/libpthread.so.0
      3216:	  trying file=/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/tls/x86_64/libpthread.so.0
      3216:	  trying file=/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/tls/libpthread.so.0
      3216:	  trying file=/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/x86_64/libpthread.so.0
      3216:	  trying file=/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/libpthread.so.0
      3216:	 search path=/opt/Qt/5.5/gcc_64/lib:/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms		(LD_LIBRARY_PATH)
      3216:	  trying file=/opt/Qt/5.5/gcc_64/lib/libc.so.6
      3216:	  trying file=/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/libc.so.6
      3216:	 search path=/opt/Qt/5.5/gcc_64/lib:/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms		(LD_LIBRARY_PATH)
      3216:	  trying file=/opt/Qt/5.5/gcc_64/lib/libdl.so.2
      3216:	  trying file=/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/libdl.so.2
      3216:	 search path=/opt/Qt/5.5/gcc_64/lib:/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms		(LD_LIBRARY_PATH)
      3216:	  trying file=/opt/Qt/5.5/gcc_64/lib/libutil.so.1
      3216:	  trying file=/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/libutil.so.1
      3216:	 search path=/opt/Qt/5.5/gcc_64/lib:/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms		(LD_LIBRARY_PATH)
      3216:	  trying file=/opt/Qt/5.5/gcc_64/lib/libexpat.so.1
      3216:	  trying file=/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/libexpat.so.1
      3216:	 search path=/opt/Qt/5.5/gcc_64/lib:/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms		(LD_LIBRARY_PATH)
      3216:	  trying file=/opt/Qt/5.5/gcc_64/lib/libz.so.1
      3216:	  trying file=/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/libz.so.1
      3216:	 search path=/opt/Qt/5.5/gcc_64/lib:/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms		(LD_LIBRARY_PATH)
      3216:	  trying file=/opt/Qt/5.5/gcc_64/lib/libm.so.6
      3216:	  trying file=/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/libm.so.6
      3216:	 search path=/opt/Qt/5.5/gcc_64/lib:/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms		(LD_LIBRARY_PATH)
      3216:	  trying file=/opt/Qt/5.5/gcc_64/lib/libgcc_s.so.1
      3216:	  trying file=/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/libgcc_s.so.1
      3216:	 search path=/opt/Qt/5.5/gcc_64/lib:/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms		(LD_LIBRARY_PATH)
      3216:	  trying file=/opt/Qt/5.5/gcc_64/lib/libresolv.so.2
      3216:	  trying file=/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/libresolv.so.2
      3216:	 search path=/opt/Qt/5.5/gcc_64/lib:/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms		(LD_LIBRARY_PATH)
      3216:	  trying file=/opt/Qt/5.5/gcc_64/lib/libbz2.so.1.0
      3216:	  trying file=/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/libbz2.so.1.0
      3216:	 search path=/opt/Qt/5.5/gcc_64/lib:/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms		(LD_LIBRARY_PATH)
      3216:	  trying file=/opt/Qt/5.5/gcc_64/lib/liblzma.so.5
      3216:	  trying file=/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/liblzma.so.5
      3216:	 search path=/opt/Qt/5.5/gcc_64/lib:/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms		(LD_LIBRARY_PATH)
      3216:	  trying file=/opt/Qt/5.5/gcc_64/lib/libnss_compat.so.2
      3216:	  trying file=/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/libnss_compat.so.2
      3216:	 search path=/opt/Qt/5.5/gcc_64/lib:/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms		(LD_LIBRARY_PATH)
      3216:	  trying file=/opt/Qt/5.5/gcc_64/lib/libnsl.so.1
      3216:	  trying file=/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/libnsl.so.1
      3216:	 search path=/opt/Qt/5.5/gcc_64/lib:/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms		(LD_LIBRARY_PATH)
      3216:	  trying file=/opt/Qt/5.5/gcc_64/lib/libnss_nis.so.2
      3216:	  trying file=/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/libnss_nis.so.2
      3216:	 search path=/opt/Qt/5.5/gcc_64/lib:/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms		(LD_LIBRARY_PATH)
      3216:	  trying file=/opt/Qt/5.5/gcc_64/lib/libnss_files.so.2
      3216:	  trying file=/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/libnss_files.so.2

asbaklm, 2017-07-10:
And my local environment setting

env | grep qt
LD_LIBRARY_PATH=/usr/local/lib/libgig:/usr/local/lib:/usr/lib/x86_64-linux-gnu:/opt/Qt/5.5/gcc_64/lib:/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms

probonopd, 2017-07-10:
Please try:
```
unset QTDIR; unset QT_PLUGIN_PATH ; unset LD_LIBRARY_PATH
./20170701_Linux_welle-io-x86_64.AppImage
```

asbaklm, 2017-07-11:
That works, thanks a lot. A previous attempt I made to remove Qt related LD_LIBRARY_PATH entries from bashrc didn't have any effect but the unset command above did the trick. Thanks! It may be worth adding this info to the FAQ / troubleshooting section.

AlbrechtL, 2017-07-11:
@probonopd Is it possible to integrate the unset of the QT paths into the AppImage?

probonopd, 2017-07-11:
The question is, why are these environment variables set in the first place on your system?

asbaklm, 2017-07-23:
I had some problems compiling and running certain packages in the past, a bunch of settings accumulated over time as different solutions were tried. It was so long ago that I can't remember what it was needed for, it may have been linuxsampler related.

closed 2017-08-02 by AlbrechtL