AlbrechtL/welle.io #92: Trouble: Windows "client" and Android "server"

lars18th, 2017-06-26:
Hi,

I'm testing Welle.io with one RTL-SDR connected to one Android smartphone, and sending the IQ data to a Windows PC client. I doesn't work reliable!

For example, the tunning frequency doesn't work. It only works when I set fixed in the driver App.
Futhermore, when I set the correct frequency in the driver, the Welle.io "client" can read the programs, but the signal is low, so no audio.

My configuration:

- RTL-SDR driver "RTL2832U driver" by Martin Marinov
- Welle.io running in a PC with the command: "welle-io.exe -D rtl_tcp -I 192.168.1.26 -P 14423"

However, if I run the "Wavesink DAB/FM Trial" in the smartphone ir works.
Any idea?

lars18th, 2017-06-27:
Hi,

The problems seems to be a poor signal.

Can I force a fixed frequency in the UI of the Welle.io?
I only have access to one DAB frequency, and a need to only scan this frenquency.

Any help?
Thank you!

AlbrechtL, 2017-06-27:
Interesting set up to use Android as a network I/Q source!  

> Can I force a fixed frequency in the UI of the Welle.io?

Yes. In the next release you can set a channel manually.  

If you are able to compile welle.io from the sources you can try it out now. The graphic design is not finished yet but the functionality is there.

[image]

closed 2017-06-27 by AlbrechtL

lars18th, 2017-06-27:
Hi @AlbrechtL ,

I can't compile it for Windows. Please, can you publish a beta?

Thank you!

AlbrechtL, 2017-06-27:
To compile it for Windows is easy. Just follow the how to https://github.com/AlbrechtL/welle.io#windows-10.

lars18th, 2017-06-28:
Hi @AlbrechtL ,

Sorry, at time I don't have the time for a full install in Window (7) of the development environment.
However, thank you for your comments. Your help is valuable!

And regarding my testing enviroment...

> Interesting set up to use Android as a network I/Q source!

It's easy to understand: Today it's very easy to obtain one smartphone (or cheap Android device) with OTG + OTG cable + RTL2832U dongle. And the RTL-SDR driver can be downloaded from the Play Store. No rooting is required and you can move the device to any place (for improve the signal reception). Then only a WIFI signal is required to send the IQ captured data to any other device in your network (PC, notebook, server, etc.).

From my point of view this is, at time, the most simple environment for newbie users! ;)