AlbrechtL/welle.io #93: Suggestion: Change the INPUT_RATE

lars18th, 2017-06-27:
Hi,

I have a very low DAB signal here... however I can listen one frequency.
But I found inconsistencies between Welle.io and Wavesink DAB/FM Trial running in the same Android device in the same position with the same driver/device/antena.

- With Wavesink running in the Android device I can listen with minimal errors the services in one frequency.
- With Welle.io I can found the services in the same frequency. However it's very difficult to listen it.

One difference between the two Apps can be the INPUT_RATE. Welle.io is using 2048000. This is defined and hardcoded in DabConstants.h It's possible to improve the signal processing increasing/reducing the rate? It's possible to include the option for changing this value?

Please, help me... as I can't receive a good signal using Welle.io.
Thank you!

AlbrechtL, 2017-06-27:
Unfortunately to change the input rate doesn't help. It can be possible that Wavesink has a better demodulator compared to welle.io. But Wavesink is closed source so we can't see what Wavesink is actually doing. 

You sad that welle.io is not working properly. Can you explain a little bit more in detail the issue, please? Did you use welle.io direct on Android or via rtl_tcp under Windows?

lars18th, 2017-06-28:
Hi @AlbrechtL ,

Thanks for your support! I'm relativelly new to softradio... ;)

I feel you're right, and perhaps the Wavesink has a better demodulator. Doing the same tests with QT-DAB with identical environment (Windows PC as rtl_tcp client) the results between Welle.io and Qt-DAB are similar. And Welle.io running in the Android device (local) and running in the PC have the same results. Then the conclusion is: Wavesink do (at time) better processing. I hope you can improve the processing in the future (if you can!).

However, I feel the input-rate can have some impact in the processing power needed to complete the demoulation. And perhaps it can have al well some effect over the resolution of thge results from the processing. Then, will be interesting to change the value and do some testing. If I'm right using the rtl_tcp protocol it's possible to select one value or other. The, why not add this option?

Regards.

AlbrechtL, 2017-06-29:
To change the input rate doesn't improve the demodulator. I recommend the book ["Digital Audio Broadcasting: Principles and Applications of DAB, DAB + and DMB"](http://eu.wiley.com/WileyCDA/WileyTitle/productCd-0470510374.html) to understand why 2048000 samples/s is OK.

closed 2017-06-29 by AlbrechtL

lars18th, 2017-06-30:
Thank you @AlbrechtL ,

I'm learning a lot about DAB with your help!