AlbrechtL/welle.io #95: No support for Zadig

Paderman, 2017-06-29:
Is it possible to run _welle.io_ with installed zadig driver instead original driver?

AlbrechtL, 2017-06-29:
Can you explain your problem a little bit more in detail, please? welle.io runs with the zadig driver, see here: https://www.welle.io/devices/rtl_sdr

Paderman, 2017-06-29:
I have a RTL2838 (R820T2, 0BDA 2838 01) stick, not RTL2832 and Zadig is installed, but welle.io say it doesn't exist a device for welle.io.

Info from Device Manager:
"Bulk-In, Interface (Interface 0)"
"Bulk-In, Interface (Interface 1)"

In Qt-DAB I must select Interface 1 (second entry in list):

[image]

andimik, 2017-06-29:
This seems to be the problem:

> RTL_SDR: Found 2 devices. Uses the first one

Could you reset the driver for Interface 0 or even install the old driver in Zadig? Not so for Interface 1, please leave it as it is.

andimik, 2017-06-29:
BTW: There is no RTL2838 at all ...

https://osmocom.org/projects/sdr/wiki/rtl-sdr

Paderman, 2017-06-29:
I try to reset the driver.

Ah, okay, a product name :)

AlbrechtL, 2017-06-30:
Is your set up working now or do you still have issues?

andimik, 2017-07-01:
Alternative: use another stick, they are very cheap, 5 Euro should be enough for 2nd hand ones.

Paderman, 2017-07-01:
It works only with Interface 1 not 0 (SDR#, Qt-DAB/SDR J DAB)

I had a stick (RT820T) but I sell it ☹ 

Hmm, then I can't use this program 😢😢😢

AlbrechtL, 2017-07-01:
If you are able to recompile welle.io you can change [this line in CRTL_SDR.spp](https://github.com/AlbrechtL/welle.io/blob/7fe91a819a80f12c68b4237105422d39d1b8b269/src/input/CRTL_SDR.cpp#L107) from  
` ret = rtlsdr_open(&device, 0);` to ` ret = rtlsdr_open(&device, 1);` to use the second interface.

Paderman, 2017-07-01:
I can try later without guarantees.

AlbrechtL, 2017-07-02:
I still don't understand why you have two devices listed but I agree that an opportunity to select the device is useful. I added it to the [task list](https://github.com/AlbrechtL/welle.io/wiki/Open-Tasks).

I recommend to start with a command line option first. In a second step some GUI functionality can be added. If you are a developer please take the job and send me a pull request!

Paderman, 2017-07-02:
Thank you for your help 🙂
I can programming minimal only in C#, PHP, but that's all...

AlbrechtL, 2017-07-02:
Great! I encourage you to learn C++ and than you can add a device index command line option for the rtl-sdr devices.

andimik, 2017-07-02:
One of my Sticks (E4000 tuner) also has two interfaces.

AlbrechtL, 2017-07-02:
Does somebody know why one dongle has two interfaces?
I mean it is obvious that in this case only the second interface works.

Paderman, 2017-07-02:
I'm very sorry, but C++ is a little to hard for me 😟 C# is a little easier, VB code is nothing for me 😂
I use VS 2017 Community and it's ok for small Windows programss or UWP...
I have a "Windows Phone" and I created a simple and not finished app (UWP)

Can you build a function to select the interface please 😇😉🤠

andimik, 2017-07-03:
I have no idea, but the E4000 PDF mentions PWM and i2C inte

jaap59, 2017-07-08:
Hi. With Windows 10, one (1) RTL-SDR v3 stick and Zadig 2.3 my laptop also shows interface 0 and 1. SDR# and HDSDR aren't bothered, but Welle-io is. Nevertheless on the same laptop using "rtl_tcp -d 1" (interface 1) and "welle-io -D rtl_tcp" (rtl_tcp interface) it works just fine. The laptop played both French and Swiss stations during my holidays close to Geneva. Good job!

AlbrechtL, 2017-08-03:
Good news! The next top priority would be to implement the support of more than one interface.

I also think the automatic selection of the interface can simply uses the rtl-sdr interface with the highest interface number. From all reports it seems that the highest interface number is the right one.

What bothers me a lot is that I don't understand why the rtl-sdr driver reports two interface even there is only one connected.

PD1LOI, 2017-08-04:
I can not recompile the file. how do you do that?

andimik, 2017-08-04:
In the meanwhile use another stick. Nowadays people are selling them for around 5 EUR

PD1LOI, 2017-08-04:
Why this stick just works on all other Win10 and Zadig 2.3 programs including SharpSdr and DDS 1.70 and so on. Only Welle.io does not work so I should look at this to solve this.

And this version I have.
[image]

HansJA, 2017-08-04:
The rtl-sdr driver interfaces are two bulk-in usb interfaces. One is the actual device and the other one is an interface for accessing the IR remote control part of the stick.
I don't know for sure what can be done with the bulk in interface 0, the IR interface, but I think you can see the IR control stream an interpret that in order to implement what the stick itself won't.

It's probably a bit adventurous to assume something like "always use i/f 1 if it's rtl2832 based"...

HansJA, 2017-08-04:
I think you should try and get in touch with other guys writing sdr utilities, and share knowledge and experience. 
SDR# and SDR-J seems to work in many ways, although sonic quality when using a 2832 stick as a wbfm stereo receiver is abysmal in both.

HansJA, 2017-08-04:
Adimik, I have two different sticks, the RTL-SDR v3 that the guy at rtl-sdr.com sells and an astrometa dvb t2 stick, and both have the same problem with "the wrong interface".

(One can't help but think of an Aardman animated film, "The Wrong Trousers"..)

AlbrechtL, 2017-08-04:
It seems that SDR# uses a modified [librtlsdr](https://github.com/librtlsdr/librtlsdr) library with [IR support](https://github.com/librtlsdr/librtlsdr/pull/9).

Somebody can try to just copy the rtlsdr.dll from SDR# to welle.io.

PD1LOI, 2017-08-05:
Please send me the ready-made file! I'm not a programmer and I do not have Linux.

Verzonden vanuit Mail voor Windows 10

Van: Albrecht Lohofener
Verzonden: vrijdag 4 augustus 2017 21:50
Aan: AlbrechtL/welle.io
CC: L Kuperij; Comment
Onderwerp: Re: [AlbrechtL/welle.io] [RTL-SDR] Support for more than one device(#95)

It seems that SDR# uses a modified librtlsdr library with IR support.
Somebody can try to just copy the rtlsdr.dll from SDR# to welle.io.

PD1LOI, 2017-08-05:
It makes no difference if I use another rtlsdr.dll. It just does not work.
I can customize CRTL_SDR.cpp with notepad ++ but I can not compile it.

Verzonden vanuit Mail voor Windows 10

Van: Albrecht Lohofener
Verzonden: vrijdag 4 augustus 2017 21:50
Aan: AlbrechtL/welle.io
CC: L Kuperij; Comment
Onderwerp: Re: [AlbrechtL/welle.io] [RTL-SDR] Support for more than one device(#95)

It seems that SDR# uses a modified librtlsdr library with IR support.
Somebody can try to just copy the rtlsdr.dll from SDR# to welle.io.

AlbrechtL, 2017-08-06:
I created a test version of welle.io with an improved auto selecting of rtl-sdr devices. It simply tries to open the first rtl-sdr device if it is not working then welle.io tries the next one and so on.
1. Please download the test version exe [here](https://gist.github.com/AlbrechtL/1fb6cc890863b21fe524b9e9ea9766d0/raw/62ee08cd9c89b969257fd7ec347fa1c0b250e0d7/20170805_rtl-sdr_welle-io.exe)
Warning: It is only the exe file of welle.io. You need to install welle.io first form [here](https://github.com/AlbrechtL/welle.io/releases/tag/V20170701).
2. Copy the exe to your welle.io install directory e.g. "C:\Program Files (x86)\welle.io"
3. Run 20170805_rtl-sdr_welle-io.exe
4. Report your results

I tested it with the welle.io version [V20170701](https://github.com/AlbrechtL/welle.io/releases/tag/V20170701).

PD1LOI, 2017-08-07:
Thank you very much, now it works well.

AlbrechtL, 2017-08-07:
I commited the code (e656145). It will be included in the official release.

Paderman, 2017-08-12:
What is if user has more than one stick on PC? 😂 😉

HansJA, 2017-08-13:
An ideal receiver application should allow selecting the receiver via a
menu.

Hans J. Albertsson
From my Nexus 5

AlbrechtL, 2017-08-13:
@HansJA Can you explain in detail the use case to use more than one device, please?

HansJA, 2017-08-14:
I usually have two different sticks connected, a commercial dvb-t2 stick
and the rtl-sdr.
In the future I envisage running an Sdrplay rsp2  in parallell with at
least one stick.

The use case is not wanting to connect +disconnect just to try the other
receiver.

Hans J. Albertsson
From my Nexus 5

Den 13 aug. 2017 20:20 skrev "Albrecht Lohofener" <notifications@github.com

AlbrechtL, 2017-08-15:
welle.io sees only dab sticks with the zadig driver or the airspy driver. I suppose that your DVB-T2 stick and the sdrplay stick uses different drivers and thus it is not a big deal to use it in parallel.

xenDE, 2019-06-28:
Hi,

is still no device_index parameter available?

like in the rtl_sdr tools:

```
$ rtl_test -h
rtl_test, a benchmark tool for RTL2832 based DVB-T receivers
Usage:
	[-d device_index (default: 0)]
$ rtl_test 
Found 2 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001
  1:  Realtek, RTL2838UHIDIR, SN: 00000001
```

best regards,
daniel

AlbrechtL, 2019-07-01:
OK guys. I reopen this issue to implement this feature in the future. It is not a really difficult task and with the version 2.0 it is easy to implement. But somebody has to do it.
At the moment our main objective is to release version 2.0. This could be something for version 2.1.

tenzap, 2019-12-08:
As a workaround, maybe you can try by using SoapySDR  which, if I remember well, allows you to choose which device you want to use thanks to the ability to define driver arguments in welle.io.

AlbrechtL, 2021-03-27:
A workaround is described here: https://github.com/AlbrechtL/welle.io/issues/615

closed 2021-03-27 by AlbrechtL